fossi-foundation/open-pdks #217: Build periodically fails with `foundry_install` error

proppy, 2022-02-15:
We setup some automated CI for `open_pdks` packaging (see https://github.com/hdl/conda-eda/issues/160).

And we're observing intermidiate failure with the following error:
```
  Extracting sky130_ef_sc_hd__fill_12 into sky130_ef_sc_hd__fill_12.ext:
  Error message output from magic script:
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__fill_4" is used but not defined in this file.
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__fill_2" is used but not defined in this file.
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__decap_6" is used but not defined in this file.
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__decap_6 was used but not defined.
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_2 was used but not defined.
  make[3]: *** [Makefile:1146: digital-hd-A] Error 1
  make[2]: *** [Makefile:940: digital-hd-build-A] Error 2
  make[1]: *** [Makefile:639: all-A] Error 2
  make: *** [Makefile:116: tech-sky130] Error 2
  Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_4 was used but not defined.
  Please name the cell before extracting.
  freeMagic called with NULL argument.
  Total of 2 warnings.
  "sky130_fd_sc_hd__decap_6" has a zero timestamp; it should be written out
      to establish a correct timestamp.
  "sky130_fd_sc_hd__fill_2" has a zero timestamp; it should be written out
      to establish a correct timestamp.
  "sky130_fd_sc_hd__fill_4" has a zero timestamp; it should be written out
      to establish a correct timestamp.
  Traceback (most recent call last):
    File "../common/foundry_install.py", line 2245, in <module>
      cwd = destlibdir,
    File "../common/foundry_install.py", line 277, in subprocess_run
      raise SystemError("".join(emsg))
  SystemError: Command magic failed with exit code: -11
```

This seems related to the following codepath in magic introduced with https://github.com/RTimothyEdwards/magic/commit/8957d4b947e6ae622c43dceda142562ee3b0e697:
https://github.com/RTimothyEdwards/magic/blob/bfdf1227a704dbbfb78f42f69965ccc2c81989c1/commands/CmdE.c#L1051
https://github.com/RTimothyEdwards/magic/blob/bfdf1227a704dbbfb78f42f69965ccc2c81989c1/commands/CmdE.c#L1017

I first though of something related to the recent parallel `make -j` support, but it turns out the error also shows up without it.

@RTimothyEdwards do you have any hint on how to dig this further?

RTimothyEdwards, 2022-02-15:
I assume that by "periodically", you mean "occasionally"?  How often are you seeing this failure?

ajelinski, 2022-02-15:
Weird here is, the successful build log also contains this "freeMagic called with NULL argument" line.

I've compared the logs from the successful and failed build jobs and I can see no really useful clues but perhaps that's just my lack of experience with `magic` (lines with `-` come from the successful build and `+` from the failed one):
```
 Starting magic under Tcl interpreter
 Using the terminal as the console.
 Using NULL graphics device.
@@ -48511,23 +48523,6 @@ Extracting sky130_fd_sc_hd__fill_4 into $SRC_DIR/sky130/sky130A/libs.ref/sky130_
 Extracting sky130_fd_sc_hd__fill_2 into $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_sc_hd/mag/sky130_fd_sc_hd__fill_2.ext:
 Extracting sky130_fd_sc_hd__decap_6 into $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_sc_hd/mag/sky130_fd_sc_hd__decap_6.ext:
 Extracting sky130_ef_sc_hd__fill_12 into sky130_ef_sc_hd__fill_12.ext:
-exttospice finished.
-Extracting cell sky130_fd_sc_hd__a2111o_1
-Extracting sky130_fd_sc_hd__a2111o_1 into sky130_fd_sc_hd__a2111o_1.ext:
-exttospice finished.
-Extracting cell sky130_fd_sc_hd__a2111oi_0
-Extracting sky130_fd_sc_hd__a2111oi_0 into sky130_fd_sc_hd__a2111oi_0.ext:
-exttospice finished.
-Extracting cell sky130_fd_sc_hd__buf_16
-Extracting sky130_fd_sc_hd__buf_16 into sky130_fd_sc_hd__buf_16.ext:
-exttospice finished.
-Extracting cell sky130_fd_sc_hd__clkdlybuf4s15_1
-Extracting sky130_fd_sc_hd__clkdlybuf4s15_1 into sky130_fd_sc_hd__clkdlybuf4s15_1.ext:
-exttospice finished.
-Extracting cell sky130_fd_sc_hd__clkdlybuf4s18_1
-Extracting sky130_fd_sc_hd__clkdlybuf4s18_1 into sky130_fd_sc_hd__clkdlybuf4s18_1.ext:
-exttospice finished.
-Done.
 Error message output from magic script:
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__fill_4" is used but not defined in this file.
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
@@ -48537,6 +48532,10 @@ Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "s
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__decap_6 was used but not defined.
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_2 was used but not defined.
+make[3]: *** [Makefile:1146: digital-hd-A] Error 1^M
+make[2]: *** [Makefile:940: digital-hd-build-A] Error 2^M
+make[1]: *** [Makefile:639: all-A] Error 2^M
+make: *** [Makefile:116: tech-sky130] Error 2^M
 Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_4 was used but not defined.
 Please name the cell before extracting.
 freeMagic called with NULL argument.
@@ -48547,19880 +48546,33 @@ Total of 2 warnings.
     to establish a correct timestamp.
 "sky130_fd_sc_hd__fill_4" has a zero timestamp; it should be written out
     to establish a correct timestamp.
-# Install all SkyWater digital standard cells.
-set -f ; ../common/foundry_install.py -std_format -timestamp 1644621075 -source $SRC_DIR/skywater-pdk/libraries -target $SRC_DIR/sky130/sky130A \
-       -techlef %l/latest/tech/*.tlef filter=custom/scripts/fix_techlefA.py \
-       -spice %l/latest/cells/*/*.spice compile-only \
-               sort=../common/sort_pdkfiles.py \
-               filter=custom/scripts/fix_device_models.py \
-       -cdl %l/latest/cells/*/*.cdl ignore=topography compile-only \
-               sort=../common/sort_pdkfiles.py \
-       -lef %l/latest/cells/*/*.lef \
-               exclude=*.magic.lef lefopts=-toplayer \
-               include=custom/sky130_fd_sc_hd/lef/sky130_ef*.lef \

...a lot more lines from the successful build...

+Traceback (most recent call last):^M
+  File "../common/foundry_install.py", line 2245, in <module>^M
+    cwd = destlibdir,^M
+  File "../common/foundry_install.py", line 277, in subprocess_run^M
+    raise SystemError("".join(emsg))^M
+SystemError: Command magic failed with exit code: -11^M
+  magic -dnull -noconsole $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_sc_hd/spice/generate_magic.tcl^M
 make[3]: Leaving directory '$SRC_DIR/sky130'
```

RTimothyEdwards, 2022-02-15:
I agree that's weird.  "freeMagic called with NULL argument" is usually the last thing magic does before it dies with a segfault.

RTimothyEdwards, 2022-02-15:
_If_ the error is related to the `sky130_ef_sc_hd__fill_12` cell, then there are simple ways to avoid the problem, although in no case should magic be generating the "freeMagic called with NULL argument" which will bother me until it's eliminated.

RTimothyEdwards, 2022-02-15:
I have prepared an alternative flattened fill cell, so we'll see if that causes the problem to go away.  I also tracked down a problem that causes "freeMagic called with NULL argument" to be printed, but it appears to be just a messaging error, not something that would cause magic to crash.  So the crash condition is still unknown.

ajelinski, 2022-03-04:
The problem with building `open_pdks` in the https://github.com/hdl/conda-eda CI seems to be gone.
I've just checked and I can't find any failing build since https://github.com/RTimothyEdwards/open_pdks/commit/1bb26d935f86dbe4a385442e73ac073cf1e1a1fe .
Thanks for fixing the problem so quickly.

closed 2022-03-07 by proppy