fossi-foundation/open-pdks #218: memory free fail after [make]

Park-Hyung-Joo, 2022-02-17:
[image]
[image]

# memory manage issue
during [make], Vmmem takes more than 25Gb of main memory and it still takes even after [sudo make install] finished

RTimothyEdwards, 2022-02-18:
I can't say what Vmmem is doing;  For Linux systems, all memory for a process is handled by the process and released when the process ends.  On my system, I do not believe that open_pdks takes more than 8GB or so, probably less, most of which is caused by the SRAM library because magic wants to read the GDS for all four SRAM macros at the same time.  If the SRAM library is not installed, then the memory overhead should be very small.  I've never tried the installation on Windows---I assume you're doing this in WSL?

Park-Hyung-Joo, 2022-02-19:
Yes, I installed them on WSL2 on Windows10. I haven't tried on Linux yet. If it has no matter on your Linux system, does it have no matter with CentOS? Actually, WSL2 is just a test case.

RTimothyEdwards, 2022-02-19:
It has never been an issue on any Linux system that I am aware of.

Park-Hyung-Joo, 2022-02-23:
I see thank you

closed 2022-02-23 by Park-Hyung-Joo