fossi-foundation/open-pdks #219: Magic Version Not Listed

donn, 2022-02-17:
There's no version of magic listed recommended to build open_pdks.

Please include this information somewhere.

donn, 2022-02-17:
Preferably in something parseable.

RTimothyEdwards, 2022-02-17:
@donn :  See `sky130/magic/sky130.tech` line 26:  `requires magic-8.3.260`.
This is the minimum version of magic required to build a usable version of open_pdks.
However, I should probably change this to 8.3.261 as there is one feature used by foundry_install that has an impact on datestamps used in GDS and LEF files, and without which the version of open_pdks is usable but the files will differ from installation to installation due to changing datestamps.

closed 2022-03-15 by donn