fossi-foundation/open-pdks #228: CI currently broken? Centos 8 is EOL?

mithro, 2022-02-28:
https://github.com/RTimothyEdwards/open_pdks/actions/runs/1909078486
```
 CentOS Linux 8 - AppStream                      237  B/s |  38  B     00:00    
  Error: Failed to download metadata for repo 'appstream': Cannot prepare internal mirrorlist: No URLs in mirrorlist
```

mithro, 2022-02-28:
@umarcor / @kguala - I think you were just doing something related to CentOS for SymbiFlow / F4PGA. If it is easy to do you help fix GitHub Actions here for @RTimothyEdwards?

mithro, 2022-02-28:
CentOS Linux 8 reached End Of Life (EOL) on December 31st, 2021 -- it seems like the next release of CentOS is "CentOS Stream 8"?

mithro, 2022-03-01:
FYI - https://www.centos.org/centos-linux-eol/

> Additionally, with this deadline falling during a time when many of our team, as well as many of our users, will be out of the office, we intend to bend the usual EOL process by keeping this content available until January 31st.

Which probably explains why only now at the end of February has things started breaking.

RTimothyEdwards, 2022-03-01:
I don't know anything about this CI, so I will definitely need help with any updates.

umarcor, 2022-03-01:
In several projects (such as Verible or f4pga-examples), CentOS 8 was removed. CentOS 7 is kept, because it is alive for at least one more year. In some places, Fedora is tested.