fossi-foundation/open-pdks #232: `Running magic to create magic database files` takes too long

donn, 2022-03-03:
This appears to be a regression in a recent patch, but essentially, `Running magic to create magic database files.` takes 3x as much time as it used to. I'm not sure how this is happening, but it's greatly inflating PDK build times.

RTimothyEdwards, 2022-03-03:
I need more information, as there is nothing updated recently that should be causing extra time here.  But if it's a real effect, I want to know what happened.

antonblanchard, 2022-04-06:
I noticed a couple of magic processes were taking quite a while to complete. Not sure if it is the same issue @donn is seeing, but a profile shows:
```
Samples: 102K of event 'cycles', Event count (approx.): 119046649203
Overhead  Command     Shared Object      Symbol
  89.35%  magicdnull  tclmagic.so        [.] DBFullResidueMask
   5.99%  magicdnull  tclmagic.so        [.] lefWriteMacro
```

antonblanchard, 2022-04-06:
A wild guess, but we did add a call to `DBFullResidueMask` in `lefWriteMacro` in https://github.com/RTimothyEdwards/magic/commit/ee1d1ae5b0219c272edd4775dccaa3e39647f381

RTimothyEdwards, 2022-04-07:
@antonblanchard :  That appears to be the result of running "lef write" on a layout with a huge number of labels, a pretty specific case.

antonblanchard, 2022-04-07:
The slow magic process is called from 'make io-A' and when I strace it, it is writing out incredible slowly (1 write every second):

```
22:29:14.845929 write(4, "END\n    PORT\n      LAYER met3 ;\n"..., 4096) = 4096
22:29:15.906640 write(4, "  RECT 67.530 198.785 67.850 199"..., 4096) = 4096
22:29:17.051336 write(4, "    END\n    PORT\n      LAYER met"..., 4096) = 4096
22:29:18.154393 write(4, "30 197.975 66.650 198.295 ;\n    "..., 4096) = 4096
22:29:19.247822 write(4, "65.900 28.060 66.100 28.260 ;\n  "..., 4096) = 4096
```

RTimothyEdwards, 2022-04-07:
It does seem to be specific to the overlay cells.  sky130_fd_io__overlay_vssio_hvc has thousands of entries for pin VSSIO.  The LEF file for that one I/O cell takes up 10% of the output LEF library.

The LEF library should not contain the overlay cells, anyway, just the top level cells.  I can change the build process to put only the top level I/O cells in the LEF library.

donn, 2022-04-07:
Then please go ahead and do that.