fossi-foundation/open-pdks #233: `config.json` lacking commit hashes for magic and skywater-pdk

donn, 2022-03-03:
There's only a commit for open_pdks, and magic, which is an empty string.

```
{
    "foundry": "SW",
    "foundry-name": "SkyWater",
    "node": "sky130A",
    "feature-size": "130nm",
    "status": "active",
    "description": "Skywater 0.13um CMOS, local interconntect + high-resistance poly + 5 metal layer backend stack + MiM caps + redistribution layer 1",
    "options": [
        "METAL5",
        "MIM",
        "REDISTRIBUTION"
    ],
    "stdcells": {
	"sky130_fd_sc_hd": "ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1",
	"sky130_fd_sc_hdll": "unknown",
	"sky130_fd_sc_hs": "unknown",
	"sky130_fd_sc_hvl": "4fd4f858d16c558a6a488b200649e909bb4dd800",
	"sky130_fd_sc_lp": "unknown",
	"sky130_fd_sc_ls": "unknown",
	"sky130_fd_sc_ms": "unknown",
	"sky130_osu_sc_t12": "unknown",
	"sky130_osu_sc_t15": "unknown",
	"sky130_osu_sc_t18": "unknown"
    },
    "iocells": {
        "sky130_fd_io": "01b18699b4102d8e54ad1406b3991ecb652e5aee"
    },
    "primitive": {
	"sky130_fd_pr": "f62031a1be9aefe902d6d54cddd6f59b57627436"
    },
    "memory": {
	"sky130_sram_macros": "unknown"
    },
    "other": {
	"sky130_ml_xx_hd": "6eb3b0718552b034f1bf1870285ff135e3fb2dcb"
    },
    "build": {
	"open_pdks": "1.0.291",
	"magic": "8.3.274"
    }
    "commit": {
	"open_pdks": "6e9a3661f5d836c2ccb89b57caec6f3dbe3e3999",
	"magic": ""
    }
}
```

donn, 2022-03-07:
@RTimothyEdwards We're blocked on this.

RTimothyEdwards, 2022-03-07:
Probably you need to update magic.  It's picking up the commit number with `magic --commit` which was implemented in revision 273 (Feb. 26).

donn, 2022-03-18:
Correct on that one, but sky130 is still missing.

RTimothyEdwards, 2022-03-18:
Why would I put a commit hash for sky130 (by which I assume you mean the skywater-pdk top level) when nothing in the open_pdks instal comes from there?  All information is pulled from the libraries, and I have a commit has for each of them.

donn, 2022-03-18:
I'll be direct, Tim. I want open_pdks to explicitly ask for a version of sky130 and magic.

This whole peer dependency situation is causing me nightmares.

donn, 2022-03-21:
The upstream skywater-pdk repo does not make any guarantees as to its structure or correctness. Both projects also lack continuous integration. Meaning at any given point in time, the only *guaranteed* working combination has to be set in OpenLane.

I'm assuming you do not test open_pdks with every possible version of Sky130 nor do you test it with Magic. So my request is simple. Whichever commit hashes (and I mean commit hashes, not versions, because upstream skywater-pdk lacks versions and the OpenLane dependency model requires consistency) of Magic and Sky130 you've *verified working* with a given commit of Open_PDKs, I would like for said values to be:

A. Included in an industry-standard marshalled format- XML, JSON or YAML, somewhere in the repository
B. The nodeinfo.json to include them as well in the "commit" field.