fossi-foundation/open-pdks #234: Broken symlinks to .mag files in install output?

mithro, 2022-03-03:
I haven't had a chance to look at this closely yet, so this is somewhat of a guess, the continous builder has been failing for ~8 days due to the following;
```
+ find /host/out/pdk-all /host/out/pdk-all.build.log /host/out/pdk-all.install.log -name '*.mag' -exec sed -i '-es/timestamp [0-9]\+/timestamp 0/' '{}' +
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_01F600.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_45.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_6D.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_41.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_64.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_7B.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_74.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_48.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_51.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2D.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_6B.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2E.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_24.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_66.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_55.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_58.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_32.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_26.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_21.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_46.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_29.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_49.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_77.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_52.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_4D.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_36.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_30.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_6F.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_3C.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_6E.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_47.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_68.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_54.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/logo_open_source_hardware.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2F.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_3B.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_79.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_63.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_22.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_7A.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_4E.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_42.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_38.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2B.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_72.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_76.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_3D.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2C.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_20.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_60.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_53.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_5C.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_50.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_75.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_4B.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_6A.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_31.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_70.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_43.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_71.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_3F.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_27.mag: No such file or directory
[... cut ...]
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_4C.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_2A.mag: No such file or directory
sed: can't read /host/out/pdk-all/usr/local/share/pdk/sky130B/libs.ref/sky130_ml_xx_hd/mag/font_69.mag: No such file or directory
```

I think the way `find` would ask `sed` to operate on a file that doesn't exist is if it was a directory or if the file was a broken symlink?

mithro, 2022-03-03:
This seems to be caused by the change at https://github.com/RTimothyEdwards/open_pdks/commit/ae8d0f9f305148f9772b4ce7f7c4bc3325068594

RTimothyEdwards, 2022-03-03:
From the output, I can figure out what happened.  The font library is basically hard-coded to sky130A, so I need to apply the same procedure I did to xschem to filter all files to turn the "sky130A" into a generic variable name that can be recast back into "sky130A" or "sky130B" as appropriate.  Should not take long to fix.

RTimothyEdwards, 2022-03-03:
I found several other places where the scripts can't distinguish between sky130A and sky130B, so fixing them all while I'm at it.

RTimothyEdwards, 2022-03-03:
This has now been fixed.  @mithro :  Please check the results after the next build.

mithro, 2022-03-03:
Looks like the build work - https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/43e103590065eb3dddcd5796fbd2d04cb510e146

mithro, 2022-03-03:
@RTimothyEdwards -- Can you look over that diff? I think the changes all are okay but I'm not smart enough to be confident in that assessment...

RTimothyEdwards, 2022-03-07:
I looked over the diff and it's okay with respect to this issue, which can be closed.  There were other issues that I'm still looking at, but they're unrelated.