fossi-foundation/open-pdks #238: Models for npn transistors not available

Unkraut3, 2022-03-17:
The models for the **npn transistors** are not listed in **all.spice**. I tried to manually add the models by adding the following lines to all.spice: 

```
.include "../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__npn_05v5_W1p00L1p00.model.spice"
.include "../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__npn_05v5_W1p00L2p00.model.spice"
```

Depending on npn model I try to use use I now get the following error from ngspice:

- 1x1µm² device: `Undefined number [dkisnpn1x1]`
- 1x2µm² device: `Undefined number [dkisnpn1x2]`

The parameters are present in the source files, e.g. `sources/sky130-pdk/libraries/sky130_fd_pr/latest/cells/npn_05v5/sky130_fd_pr__npn_05v5__t.corner.spice` but apparently are not transferred during conversion.

Adding to the following to my spice file before including all.spice enables to perform simulations using typical values:
```
.param
+ dkisnpn1x1=8.7913e-01 dkbfnpn1x1=9.8501e-01
+ dkisnpn1x2=9.0950e-01 dkbfnpn1x2=9.6759e-01
+ dkisnpnpolyhv=1.0 dkbfnpnpolyhv=1.0
```

It would be great if the npn transistor models would be available for ngspice.

RTimothyEdwards, 2022-03-17:
(1) Yes, the NPN was missing from the original SkyWater all.spice file.  I have a script in open_pdks that is supposed to fix this.  It mentions that PNP and NPN includes are missing, and then it fixes the PNPs but not the NPNs.  I just corrected this and pushed (to the opencircuitdesign.com repository).

(2) What you want to be doing (after applying the fix to "all.spice") in your testbench SPICE is `.lib /path/to/sky130.lib.spice corner` where `corner` is `tt`, `ff`, etc., and includes options for BEOL corners, mismatch analysis, and monte carlo simulation.  The `sky130.lib.spice` file includes the corner models, so you don't need to specify the corner parameters yourself.

Unkraut3, 2022-03-29:
Thank you @RTimothyEdwards. In the current version of open_pdks the transistor models are now included, but the parameters  (.param dkisnpn1x1, dkbfnpn1x1, dkisnpn1x2, dkbfnpn1x2, dkisnpnpolyhv, dkbfnpnpolyhv,) are not set. I still need to set e.g. 

```
.param
+ dkisnpn1x1=8.7913e-01 dkbfnpn1x1=9.8501e-01
+ dkisnpn1x2=9.0950e-01 dkbfnpn1x2=9.6759e-01
+ dkisnpnpolyhv=1.0 dkbfnpnpolyhv=1.0
```

I've attached spice files that should run through and create some plots if the transistor models are available: [esd_clamp.zip](https://github.com/RTimothyEdwards/open_pdks/files/8372729/esd_clamp.zip)