fossi-foundation/open-pdks #239: Use spef_extractor OpenRCX rulesets instead of magic RCX rulesets

donn, 2022-03-18:
The magic RCX rulesets are showing timing violations: https://github.com/The-OpenROAD-Project/OpenLane/actions/runs/2000464858

According to @mkkassem, these are inaccurate. The new ones should be installed in their place, which are available at https://github.com/efabless/openrcx-calibration/tree/main/openrcx-spef_extractor for both sky130A and sky130B.

RTimothyEdwards, 2022-03-18:
Did you use the the new RCX calibration files that were updated in open_pdks around the same time as your post?

RTimothyEdwards, 2022-03-18:
I tried a synthesis run yesterday with the files from yesterday and openlane failed with timing errors.  I tried it again today with the openRCX calibration files I generated from magic yesterday after modifying the way that magic calculates parasitics, and openlane passed (well, it died on something obscure, but seems to have gotten through timing analysis).

The difference is that yesterday I came across metal parasitic coefficients that I didn't know about that were buried in the model files.  I took some time to understand the format, and then came to realize that there is a huge effect of proximal shapes shielding fringe capacitance that magic did not calculate.  That caused magic to way overestimate fringe capacitance on closely spaced wires----i.e., everything in a routed standard cell layout.  I also realized that it was pretty easy to model the effect analytically, and for magic to calculate the shielding and apply it to all fringe calculations.  So I was able to update magic's capacitance calculations yesterday, and then rewrote new calibration files for openRCX.

There are some minor 2nd order effects that I might need to add, namely the dependence of fringe capacitance magnitude on wire width, but that is a relatively minor adjustment compared to the fringe shielding effect.

donn, 2022-03-19:
I used 4040, so yes.

RTimothyEdwards, 2022-03-19:
4040 was from March 14, so no.  You would need `6427317204abdc44c274cc7799cfc250843e10db`.

RTimothyEdwards, 2022-03-19:
Also, I discovered parasitic capacitance models at corners buried in the SPICE models, so I spent pretty much all yesterday building out the extraction deck in magic for minimum and maximum corners, so I plan to do another update of the calibration values later today, if you want to wait for that.  The nominal values will not change.

donn, 2022-03-19:
Something may be up with your GitHub sync, then, because 6427317204abdc44c274cc7799cfc250843e10db is the latest I see.

RTimothyEdwards, 2022-03-19:
Ugh, you're right.  Thanks for the heads-up.

RTimothyEdwards, 2022-03-19:
It appears that github did some update a few days ago and now I can no longer access it with ssh from my server machine (which is pretty old).  I will see if I can run the mirror from another of my machines.

RTimothyEdwards, 2022-03-19:
Okay, I ran the mirror copy from my desktop, so `open_pdks` and `magic` are now updated on github.  Note that the change in the way magic handles parasitics required a new keyword for the tech file, so there is now a dependency in open_pdks on the most recent version of magic.