fossi-foundation/open-pdks #241: Hspice/Spectre model

lekez2005, 2022-03-23:
I have developed local python scripts that modify the supplied ngspice models to support hspice/spectre (aka much much faster) simulations. Would you be potentially interested in integrating them into open_pdks? If so, I can work on a pull request and then you can decide. I see from the comments in `fixspice.py` that a more robust model is on the way so I suppose this may not be needed but asking just in case.

lekez2005, 2022-03-23:
Alright, I'll send a pull request later today

mattvlsi, 2022-04-18:
Hi @lekez2005. Where can I find this script and how should I run it? Can you explain please? Thanks.

lekez2005, 2022-04-18:
I have a pull request here: https://github.com/RTimothyEdwards/open_pdks/pull/244/. But really, the hspice branch in my fork should also work for building it: https://github.com/lekez2005/open_pdks/tree/hspice. If you don't care about building for yourself, then you can download and extract [hspice-model.tar.bz2](https://github.com/lekez2005/open_pdks/releases/download/1.0.306/hspice-model.tar.bz2) at https://github.com/lekez2005/open_pdks/releases