fossi-foundation/open-pdks #246: Add gate level simulation verilog to PDK.

donnie-j, 2022-03-31:
I needed to simulate the gate level netlist for DFFRAM, but I found that no combination of defines results in functional results with icarus verilog.  Defining FUNCTIONAL gets close, but it seems to pull in some behavioural models, which are broken for simulation.  Also, the lpflow and spare cell macros cause problems, so they need to be excluded (or fixed).

I went back to the SkyWater sources to find the correct models to include, and this is the 'recipe' to create a pure simulation verilog library:

```
$ (for i in `ls | grep -v lpflow | grep -v macro ` ; do find $i -name \*functional\*v | grep -v \.pp\. ; find $i -name \*_[0-9]*.v  ; done) > lst
$ echo \`define UNIT_DELAY \#1 > sky130_fd_sc_hd_sim.v
$ cat $PDK_ROOT/sky130A/libs.ref/sky130_fd_sc_hd/verilog/primitives.v >> sky130_fd_sc_hd_sim.v
$ for i in `cat lst` ; do grep -v ^\`include $i >> sky130_fd_sc_hd_sim.v; echo >> sky130_fd_sc_hd_sim.v; done
```

This file can then be used very simply to run post synthesis simulations, e.g.:

```
$ iverilog -o ram_tb sky130_fd_sc_hd_sim.v RAM8.nl.v ram_tb.v ; vvp ram_tb
VCD info: dumpfile ram_tb.vcd opened for output.
```

RTimothyEdwards, 2022-03-31:
I've been doing gate level simulations for a long time now without issues.  Possibly this is because you are doing simulations without power pins?  These lines are in the Caravel project verilog and gate level simulation works:
```
        `define UNIT_DELAY #1
        `define USE_POWER_PINS
```
And then include the file of verilog primitives and the library.  Possibly the problem is in the definition of files without power pins?

donnie-j, 2022-03-31:
I think there are 2 different (possibly related) problems here.

The first is that the lpflow and to a lesser extent spare macro cells don't really make much sense without the power pins because their functionality depends somewhat on what the power rails do.  Since they've been included in the single verilog library, not setting `USE_POWER_PINS` causes them to throw parse errors with iverilog.  I'm kinda Verilog as a Second Language, and I don't immediately see what is causing that.  The simple work around might be to `` `ifdef`` those models out when power pins are not used, which would at least point the user in the right direction.

The second problem I think is with the behavioural version of `dlclkp` in `sky130_fd_sc_hd__dlclkp.behavioral.v`.  It has a signal `GATE_delayed` which AFAICS is never assigned.  The `.functional.v` version of this model works properly.  The result is that the Open_PDK version of these models, even with power pins, never passes the clock though the instances of this gate in the DFFRAM designs (that I've run).

Here's what it's supposed to look like, and with the powered Open_PDK model of `dlclkp`
[image]
[image]

I've attached the powered and unpowered models and test benches here also
[ram_tb.tar.gz](https://github.com/RTimothyEdwards/open_pdks/files/8393222/ram_tb.tar.gz)
.

mattvlsi, 2022-04-18:
Hi @donnie-j and @RTimothyEdwards. I'm facing same issue. When **`define USE_POWER_PINS** commented out, iverilog gives some syntax errors.  Where should I run the code recipe above to have a sky130_fd_sc_hd_sim.v properly? Because the first row of code searchs some files and seperates, I guess. Thanks.

donnie-j, 2022-04-19:
Hi @mattvlsi. I found I needed to go back to the https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd sources.  IIRC, you want to run that in the cells directory... but I'm not suggesting that that is what ppl should do, just to show the quick hack that produced a working library.

mattvlsi, 2022-04-19:
Thank you very much @donnie-j. It worked for now.