fossi-foundation/open-pdks #251: New RCX tables show significant increase in parasitic capacitances

antonblanchard, 2022-04-08:
I'm seeing a 2-3x increase in parasitic capacitances when testing the old RCX tables vs the new ones. I know the old ones are suspect, but I just wanted to make sure this large an increase was expected.

If the new ones are correct, we will need to update the capacitance estimates the resizer uses, so it has a better chance at fixing setup and hold violations.

Test case is here: [rcx-old-vs-new.tar.gz](https://github.com/RTimothyEdwards/open_pdks/files/8448836/rcx-old-vs-new.tar.gz). rcx.tcl creates the SPEF files, sta.tcl does the timing analysis.

In the test case I dump one of the longest paths which goes from 8.53 ns  to 13.91 ns with the new tables:

```
Startpoint: _617_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _357_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock source latency
                  0.39    0.29    0.29 ^ clk (in)
     1    0.09                           clk (net)
                  0.39    0.00    0.29 ^ clkbuf_0_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.10    0.28    0.57 ^ clkbuf_0_clk/X (sky130_fd_sc_hd__clkbuf_16)
     4    0.08                           clknet_0_clk (net)
                  0.10    0.00    0.57 ^ clkbuf_2_2_0_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.22    0.29    0.86 ^ clkbuf_2_2_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
     4    0.13                           clknet_2_2_0_clk (net)
                  0.22    0.01    0.87 ^ clkbuf_4_10__f_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.15    0.28    1.15 ^ clkbuf_4_10__f_clk/X (sky130_fd_sc_hd__clkbuf_16)
    10    0.14                           clknet_4_10__leaf_clk (net)
                  0.15    0.01    1.17 ^ clkbuf_leaf_98_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.03    0.16    1.32 ^ clkbuf_leaf_98_clk/X (sky130_fd_sc_hd__clkbuf_16)
     2    0.01                           clknet_leaf_98_clk (net)
                  0.03    0.00    1.32 ^ _617_/CLK (sky130_fd_sc_hd__dfxtp_2)
                  0.34    0.54    1.87 ^ _617_/Q (sky130_fd_sc_hd__dfxtp_2)
     3    0.07                           a_registered[1] (net)
                  0.34    0.01    1.88 ^ repeater1205/A (sky130_fd_sc_hd__buf_4)
                  0.17    0.30    2.18 ^ repeater1205/X (sky130_fd_sc_hd__buf_4)
     9    0.06                           net1205 (net)
                  0.17    0.01    2.18 ^ U$$3/A (sky130_fd_sc_hd__and2_1)
                  0.05    0.16    2.34 ^ U$$3/X (sky130_fd_sc_hd__and2_1)
     1    0.00                           t (net)
                  0.05    0.00    2.34 ^ U$$4/B1 (sky130_fd_sc_hd__a32o_1)
                  0.29    0.31    2.64 ^ U$$4/X (sky130_fd_sc_hd__a32o_1)
     4    0.03                           sel_0 (net)
                  0.29    0.00    2.64 ^ repeater447/A (sky130_fd_sc_hd__buf_4)
                  0.22    0.32    2.97 ^ repeater447/X (sky130_fd_sc_hd__buf_4)
     9    0.08                           net447 (net)
                  0.22    0.01    2.98 ^ repeater445/A (sky130_fd_sc_hd__clkbuf_4)
                  0.20    0.32    3.30 ^ repeater445/X (sky130_fd_sc_hd__clkbuf_4)
     9    0.06                           net445 (net)
                  0.20    0.01    3.30 ^ repeater443/A (sky130_fd_sc_hd__buf_4)
                  0.29    0.34    3.65 ^ repeater443/X (sky130_fd_sc_hd__buf_4)
     9    0.11                           net443 (net)
                  0.29    0.02    3.67 ^ repeater442/A (sky130_fd_sc_hd__buf_4)
                  0.15    0.28    3.94 ^ repeater442/X (sky130_fd_sc_hd__buf_4)
     7    0.05                           net442 (net)
                  0.15    0.00    3.94 ^ U$$120/A2 (sky130_fd_sc_hd__a22o_1)
                  0.07    0.18    4.13 ^ U$$120/X (sky130_fd_sc_hd__a22o_1)
     1    0.01                           t$4498 (net)
                  0.07    0.00    4.13 ^ U$$121/A (sky130_fd_sc_hd__xor2_1)
                  0.08    0.09    4.22 v U$$121/X (sky130_fd_sc_hd__xor2_1)
     1    0.01                           booth_b0_m57 (net)
                  0.08    0.00    4.22 v dadda_fa_0_57_0/A (sky130_fd_sc_hd__fa_1)
                  0.10    0.48    4.70 v dadda_fa_0_57_0/COUT (sky130_fd_sc_hd__fa_1)
     1    0.01                           c$18 (net)
                  0.10    0.00    4.70 v dadda_fa_1_58_7/A (sky130_fd_sc_hd__fa_1)
                  0.13    0.53    5.24 v dadda_fa_1_58_7/COUT (sky130_fd_sc_hd__fa_1)
     1    0.01                           c$512 (net)
                  0.13    0.00    5.24 v dadda_fa_2_59_2/CIN (sky130_fd_sc_hd__fa_1)
                  0.10    0.51    5.75 v dadda_fa_2_59_2/SUM (sky130_fd_sc_hd__fa_1)
[... 78 lines cut ...]
     1    0.01                           c$18 (net)
                  0.11    0.00    8.89 v dadda_fa_1_58_7/A (sky130_fd_sc_hd__fa_1)
                  0.25    0.67    9.57 v dadda_fa_1_58_7/COUT (sky130_fd_sc_hd__fa_1)
     1    0.04                           c$512 (net)
                  0.25    0.00    9.57 v dadda_fa_2_59_2/CIN (sky130_fd_sc_hd__fa_1)
                  0.11    0.58   10.15 v dadda_fa_2_59_2/SUM (sky130_fd_sc_hd__fa_1)
     1    0.01                           s$1454 (net)
                  0.11    0.00   10.15 v dadda_fa_3_59_3/A (sky130_fd_sc_hd__fa_1)
                  0.24    0.72   10.86 v dadda_fa_3_59_3/SUM (sky130_fd_sc_hd__fa_1)
     1    0.04                           s$2338 (net)
                  0.24    0.00   10.87 v dadda_fa_4_59_2/CIN (sky130_fd_sc_hd__fa_1)
                  0.24    0.66   11.53 v dadda_fa_4_59_2/COUT (sky130_fd_sc_hd__fa_1)
     1    0.04                           c$3073 (net)
                  0.24    0.00   11.53 v dadda_fa_5_60_0/CIN (sky130_fd_sc_hd__fa_2)
                  0.24    0.64   12.17 v dadda_fa_5_60_0/COUT (sky130_fd_sc_hd__fa_2)
     1    0.08                           c$3664 (net)
                  0.24    0.00   12.17 v dadda_fa_6_61_0/A (sky130_fd_sc_hd__fa_1)
                  0.69    1.11   13.29 ^ dadda_fa_6_61_0/SUM (sky130_fd_sc_hd__fa_1)
     1    0.07                           s$4049 (net)
                  0.69    0.01   13.29 ^ dadda_fa_7_61_0/CIN (sky130_fd_sc_hd__fa_1)
                  0.12    0.61   13.91 v dadda_fa_7_61_0/SUM (sky130_fd_sc_hd__fa_1)
     1    0.01                           s$4306 (net)
                  0.12    0.00   13.91 v _357_/D (sky130_fd_sc_hd__dfxtp_1)
                                 13.91   data arrival time
```

antonblanchard, 2022-04-08:
I used @jjcherry56 scripts from ORFS to recalculate the layer capacitance estimates (in pF/micron of wire length). The calculated estimates are about 3x bigger than the LAYER_RC values we were using until recently, more or less matching what the STA data above suggested. All layers are off (this was a macro so there was no met5):

```
       met1 1.449e-04 -> 7.15469e-04 
       met2 1.331e-04 -> 5.54684e-04 
       met3 1.464e-04 -> 6.89480e-04 
       met4 1.297e-04 -> 3.96776e-04 
```

I checked the TLEF capacitance values, and they appear to be even smaller than the recent LAYER_RC values. eg for met1:

```
WIDTH 0 0.14
EDGECAPACITANCE 40.567E-6 ;
CAPACITANCE CPERSQDIST 25.7784E-6 ;
```

```
cap_pf_per_micron = layer_width * layer->getCapacitance() + 2 * layer->getEdgeCapacitance();
```

ie `8.5e-5 pF/micron`.

antonblanchard, 2022-04-08:
https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=226894802

I'm not sure if this sky130 data is accurate, but it suggests to me a (worst case) min width (0.14), min spacing (0.14) wire has a total capacitance of `3.3e-04 pF/um`, way below the `7.15e-04 pF/um` estimated above from the new RCX tables.

It also suggests the TLEF values are way too low. I'm a bit lost as to what is correct.

RTimothyEdwards, 2022-04-08:
"I'm a bit lost as to what is correct."---Join the club!

I was also trying to use FasterCap to get a field equation solver's take on the values, but I have not been able to get meaningful results out of FasterCap (James Stine has some students who apparently were able to get results from some field solver, so I need to get in touch with them to figure out what I might be doing wrong).

Something is clearly off here.  I have always been able to cross-check the area capacitance, which is simply Cox/tox.  The fringe capacitance is hard to get a theoretically "correct" value for.  But for metal1, Cox/tox gives 25.78 aF/um^2, which is the same as the value in TLEF.  That value times the line width of 0.14um gives 3.6 aF/um.  But the table you cited the link to above has 35.4 aF/um, which is demonstrably off by a factor of 10.  Where does that table come from?  There is nothing on the page to suggest where it comes from or why it should be trusted.

It has never been clear to me what the CPERSQDIST in a LEF file represents.  Presumably it's the maximum fringe capacitance if there are no nearby wires, but that would make it basically useless for calculating fringe of a wire that is closely spaced to other wires.  In the best case, I would expect capacitance values calculated only from the TLEF values to be over-estimates.

RTimothyEdwards, 2022-04-08:
Is there somewhere I can find documentation on what the numbers are in the OpenRCX input tables?

antonblanchard, 2022-04-08:
> Is there somewhere I can find documentation on what the numbers are in the OpenRCX input tables?

It was linked from https://skywater-pdk.readthedocs.io/en/main/rules/rcx.html#capacitance-values . Perhaps @mithro knows where it came from?

mithro, 2022-04-08:
There is a spreadsheet with some more data @ https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=1391862914

RTimothyEdwards, 2022-04-08:
@mithro :  That's not really helpful because neither table says anything about where the data came from.  It's like a table that appeared out of nowhere.

mithro, 2022-04-08:
Doesn't the page @ https://skywater-pdk.readthedocs.io/en/main/rules/rcx.html

> This table list layers and contacts included in SPICE models, and parasitic layers include in the AssuraLayout Extraction.
> 
> The modeled columns indicate sheets and contacts that are parasitic resistance/capacitance included in the model extraction measurements.
> 
> The CAD columns indicate sheets and contacts that are parasitics included in the schematic/layout RCX from Assura.

RTimothyEdwards, 2022-04-09:
@mithro :  The specific part that you quoted refers to parasitics that are included in SPICE device models.

From what I see further down on that page, there are resistance and capacitance values that were taken from the tables I created in another spreadsheet and have nothing to do whatsoever with the tables that are pointed to by the links.

antonblanchard, 2022-04-18:
I reran the same test with the recent RCX updates in open_pdks:

```
Startpoint: _617_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _357_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock source latency
                  0.24    0.18    0.18 ^ clk (in)
     1    0.05                           clk (net)
                  0.24    0.00    0.18 ^ clkbuf_0_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.07    0.22    0.40 ^ clkbuf_0_clk/X (sky130_fd_sc_hd__clkbuf_16)
     4    0.04                           clknet_0_clk (net)
                  0.07    0.00    0.40 ^ clkbuf_2_2_0_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.13    0.21    0.61 ^ clkbuf_2_2_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
     4    0.07                           clknet_2_2_0_clk (net)
                  0.13    0.01    0.61 ^ clkbuf_4_10__f_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.12    0.23    0.84 ^ clkbuf_4_10__f_clk/X (sky130_fd_sc_hd__clkbuf_16)
    10    0.11                           clknet_4_10__leaf_clk (net)
                  0.12    0.01    0.85 ^ clkbuf_leaf_98_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.03    0.15    1.00 ^ clkbuf_leaf_98_clk/X (sky130_fd_sc_hd__clkbuf_16)
     2    0.00                           clknet_leaf_98_clk (net)
                  0.03    0.00    1.00 ^ _617_/CLK (sky130_fd_sc_hd__dfxtp_2)
                  0.09    0.37    1.37 v _617_/Q (sky130_fd_sc_hd__dfxtp_2)
     3    0.03                           a_registered[1] (net)
                  0.09    0.01    1.38 v repeater1205/A (sky130_fd_sc_hd__buf_4)
                  0.07    0.20    1.58 v repeater1205/X (sky130_fd_sc_hd__buf_4)
     9    0.04                           net1205 (net)
                  0.07    0.00    1.58 v U$$3/A (sky130_fd_sc_hd__and2_1)
                  0.03    0.15    1.73 v U$$3/X (sky130_fd_sc_hd__and2_1)
     1    0.00                           t (net)
                  0.03    0.00    1.73 v U$$4/B1 (sky130_fd_sc_hd__a32o_1)
                  0.10    0.27    2.00 v U$$4/X (sky130_fd_sc_hd__a32o_1)
     4    0.02                           sel_0 (net)
                  0.10    0.00    2.00 v repeater447/A (sky130_fd_sc_hd__buf_4)
                  0.07    0.21    2.21 v repeater447/X (sky130_fd_sc_hd__buf_4)
     9    0.04                           net447 (net)
                  0.07    0.01    2.22 v repeater445/A (sky130_fd_sc_hd__clkbuf_4)
                  0.09    0.21    2.43 v repeater445/X (sky130_fd_sc_hd__clkbuf_4)
     9    0.04                           net445 (net)
                  0.09    0.00    2.44 v repeater443/A (sky130_fd_sc_hd__buf_4)
                  0.08    0.22    2.65 v repeater443/X (sky130_fd_sc_hd__buf_4)
     9    0.05                           net443 (net)
                  0.08    0.01    2.66 v repeater442/A (sky130_fd_sc_hd__buf_4)
                  0.06    0.19    2.85 v repeater442/X (sky130_fd_sc_hd__buf_4)
     7    0.03                           net442 (net)
                  0.06    0.00    2.85 v U$$120/A2 (sky130_fd_sc_hd__a22o_1)
                  0.05    0.25    3.11 v U$$120/X (sky130_fd_sc_hd__a22o_1)
     1    0.00                           t$4498 (net)
                  0.05    0.00    3.11 v U$$121/A (sky130_fd_sc_hd__xor2_1)
                  0.07    0.18    3.29 v U$$121/X (sky130_fd_sc_hd__xor2_1)
     1    0.01                           booth_b0_m57 (net)
                  0.07    0.00    3.29 v dadda_fa_0_57_0/A (sky130_fd_sc_hd__fa_1)
                  0.09    0.47    3.76 v dadda_fa_0_57_0/COUT (sky130_fd_sc_hd__fa_1)
     1    0.01                           c$18 (net)
                  0.09    0.00    3.76 v dadda_fa_1_58_7/A (sky130_fd_sc_hd__fa_1)
                  0.10    0.50    4.26 v dadda_fa_1_58_7/COUT (sky130_fd_sc_hd__fa_1)
     1    0.01                           c$512 (net)
                  0.10    0.00    4.26 v dadda_fa_2_59_2/CIN (sky130_fd_sc_hd__fa_1)
                  0.10    0.50    4.76 v dadda_fa_2_59_2/SUM (sky130_fd_sc_hd__fa_1)
     1    0.01                           s$1454 (net)
                  0.10    0.00    4.76 v dadda_fa_3_59_3/A (sky130_fd_sc_hd__fa_1)
                  0.11    0.55    5.31 v dadda_fa_3_59_3/SUM (sky130_fd_sc_hd__fa_1)
     1    0.01                           s$2338 (net)
                  0.11    0.00    5.31 v dadda_fa_4_59_2/CIN (sky130_fd_sc_hd__fa_1)
                  0.11    0.46    5.77 v dadda_fa_4_59_2/COUT (sky130_fd_sc_hd__fa_1)
     1    0.01                           c$3073 (net)
                  0.11    0.00    5.77 v dadda_fa_5_60_0/CIN (sky130_fd_sc_hd__fa_2)
                  0.10    0.43    6.20 v dadda_fa_5_60_0/COUT (sky130_fd_sc_hd__fa_2)
     1    0.02                           c$3664 (net)
                  0.10    0.00    6.20 v dadda_fa_6_61_0/A (sky130_fd_sc_hd__fa_1)
                  0.14    0.65    6.85 ^ dadda_fa_6_61_0/SUM (sky130_fd_sc_hd__fa_1)
     1    0.01                           s$4049 (net)
                  0.14    0.00    6.85 ^ dadda_fa_7_61_0/CIN (sky130_fd_sc_hd__fa_1)
                  0.08    0.45    7.30 v dadda_fa_7_61_0/SUM (sky130_fd_sc_hd__fa_1)
     1    0.00                           s$4306 (net)
                  0.08    0.00    7.30 v _357_/D (sky130_fd_sc_hd__dfxtp_1)
                                  7.30   data arrival time
```

7.3 ns - faster than the previous two versions (8.53 ns to 13.91 ns).

antonblanchard, 2022-04-18:
I also made an attempt to match the TLEF capacitance values to the SPEF results. Taking the TLEF data and converting it to fF/um for a min width wire, I get:

```
li1 8.7682e-2 fF/um
met1 8.4743e-2 fF/um
met2 7.7890e-2 fF/um
met3 8.5690e-2 fF/um
met4 7.5877e-2 fF/um
met5 8.7815e-2 fF/um
```

I compared this to the output of `correlateRC.py` from ORFS, which calculates updated layer capacitance estimates. On user_proj_example, and three of my designs from caravel:

user_proj_example
```
Coefficient of determination: 0.9800
New layer capacitances:
Layer met1: 7.84240E-02 fF/um
Layer met2: 7.45669E-02 fF/um
Layer met3: 8.37758E-02 fF/um
Layer met4: 6.79889E-02 fF/um
```
multiply_add_64x64
```
Coefficient of determination: 0.9928
New layer capacitances:
Layer met1: 8.24512E-02 fF/um
Layer met2: 7.64900E-02 fF/um
Layer met3: 6.52173E-02 fF/um
Layer met4: 7.60617E-02 fF/um
```
Microwatt_FP_DFFRFile
```
Coefficient of determination: 0.9971
New layer capacitances:
Layer met1: 8.37255E-02 fF/um
Layer met2: 7.94404E-02 fF/um
Layer met3: 7.58108E-02 fF/um
Layer met4: 7.54418E-02 fF/um
```
user_project_wrapper
```
Coefficient of determination: 0.9802
New layer capacitances:
Layer met1: 7.17819E-02 fF/um
Layer met2: 8.17310E-02 fF/um
Layer met3: 8.39879E-02 fF/um
Layer met4: 7.69995E-02 fF/um
Layer met5: 9.09862E-02 fF/um
```

So the TLEF data and the SPEF data are for the most part with 10% of each other which is good. I also notice the timing report after global routing is now much closer to the timing report after SPEF extraction which is good. For one thing it means our hold violation fixups are much more accurate, and I notice I don't need to over fix my designs as much as I used to.

mithro, 2022-05-11:
Hi @antonblanchard, 

Your last comment seems to say everything is good? Shall we close this issue now?

Tim 'mithro' Ansell

maliberty, 2022-05-11:
I believe there are two issues here

- RCX having " 2-3x increase"  - still open and critical
- GRT correlation to RCX - seemingly resolved

antonblanchard, 2022-05-11:
I agree with @maliberty. The open issue is that we now have 3 sets of RCX data, and it sounds like only the OpenROAD RCX data has been correlated to known good data. The Openlane RCX data has not.

mithro, 2022-05-11:
3 sets?

antonblanchard, 2022-05-11:
OpenROAD, RCX calculated from magic (which I think was the 3x delta) and current Openlane (which doesn't match OpenROAD), at least with my testing.

maliberty, 2022-05-11:
It appears a third set based on "spef_extractor" was generated - I don't know the details.  @RTimothyEdwards @donn can you explain further?