fossi-foundation/open-pdks #255: Gate Level Simulation

mattvlsi, 2022-04-19:
I want to do gate level simulation with sky130_fd_sc_hd.v and primitives.v without using power nets. However iverilog can't compile sky130_fd_sc_hd.vwithout power nets because of some cells syntax error. How can I work around this? Thanks.

RTimothyEdwards, 2022-04-19:
I do gate level simulation with power nets all the time.  You need to be more specific about the syntax error.  I am aware of a couple of cells with errors but usually this is just a matter of using the functional rather than the behavioral model (I will eventually get around to fixing those errors, which are in the skywater-pdk; since there is a workaround, it has been a lower priority for me).

mattvlsi, 2022-04-19:
Hi @RTimothyEdwards. I faced this issue during trying to simulate a post-synthesis netlist simulation that is generated by Yosys in OpenLane. And the netlist doesn't have power nets. Am I missing some point? Thanks.