fossi-foundation/open-pdks #258: LEF/DEF with DRC errors generate GDS without any

donn, 2022-06-23:
Based on The-OpenROAD-Project/OpenLane#695.

---

A tlef/def combo reports DRC errors when run with Magic, but running DRC on said GDS file returns 0 errors.

* [reproducible.tar.gz](https://github.com/RTimothyEdwards/open_pdks/files/8966633/reproducible.tar.gz)

RTimothyEdwards, 2022-06-23:
@donn :
Magic interprets the BLOCKAGE as an obstruction and places the obstruction over the entire area specified by the BLOCKAGE statement.  It is not clear to me (from the LEF/DEF documentation) how a layer BLOCKAGE is supposed to be differentiated from an OBS layer obstruction in a LEF file, but that is how magic interprets it.

Would it be helpful if I added an option to "def read" to ignore any BLOCKAGE sections in the input?

RTimothyEdwards, 2022-06-23:
Yes, that's exactly right.  There is no OBS statement for DEF files, so BLOCKAGE is treated as such.  The obstruction is assumed to be in the position of an actual piece of metal somewhere in the design.  Magic's DRC treats obstructions like metal for the purpose of DRC checks.

RTimothyEdwards, 2022-06-23:
I do think that the proper way to handle it is to have an option to ignore blockages when reading DEF into magic.

donn, 2022-06-23:
Then yeah, let's do that.

donn, 2022-06-23:
…I just realized I opened this in open_pdks, not magic. Welp.

RTimothyEdwards, 2022-06-23:
@donn:  I have implemented this in magic as `def read <filename> -noblockage` and tested it on your example;  it works as expected.

Because I am upgrading my Open Circuit Design server machine, and I cloned my repository off that machine, I am not going to push the update immediately, but I hope to get it pushed by end of day today.