fossi-foundation/open-pdks #268: Liberty files from GF180MCU have no headers

donn, 2022-08-10:
* `gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_1v80.lib`:

```
/*
 * Copyright 2022 GlobalFoundries PDK Authors
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

  cell(gf180mcu_fd_sc_mcu7t5v0__addf_1) {
    area : 72.441600 ;
```

Note only does the library not have a name, making it unreadable by most liberty file parsers, hacking in a name does not set the `time_unit` and the `capacitive_load_unit`.

* Log: [install.log.gz](https://github.com/RTimothyEdwards/open_pdks/files/9302721/install.log.gz)

RTimothyEdwards, 2022-08-22:
@donn :  This should be taken up in the issue tracker for the GF open PDK repository.  The original GF liberty files were broken up into individual cells, and apparently the library header was discarded because I cannot find any header in the repository.  There is no "make timing" like there is in the SkyWater repository, which uses a script `scripts/skywater_pdk/liberty.py` to generate the header from scratch.  There is no `scripts` directory in GF at all.  A collection of the missing headers put in, say, `globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/tech` would be sufficient, and, frankly, a lot simpler than running a python script.

RTimothyEdwards, 2022-08-22:
In short, I have to go back to the original GF files to be able to reproduce the headers, because those header are not in the open source repository.  I will still need to make a change to open_pdks to handle the header in the liberty library.  The script `create_lib_library.py` has a caveat in the code:  `Warning:  This script is unfinished.  Needs to parse the library header. . .`.

donn, 2022-08-22:
@mithro Weigh in?

mithro, 2022-08-22:
?

RTimothyEdwards, 2022-08-22:
@mithro :  Is this not clear?  The SkyWater liberty files have the header inserted by a python script when doing "make timing".  There is no such python script in the GF repository, no "make timing" recipe, therefore no header recreated.  I would prefer that the headers just be put somewhere rather than be regenerated by script, but it's your choice whether to reconstruct the liberty libraries within the PDK repository, or let open_pdks do it.  Regardless, the per-cell liberty file fragments are meaningless without the header information, as @donn points out.

mithro, 2022-08-23:
Create a GitHub Issue on the GF PDK. I agree there should be a `make timing` equivalent.

RTimothyEdwards, 2022-08-23:
@mithro :  Done.