fossi-foundation/open-pdks #270: [Minor issue] CLASS CORE SPACER missing for some spacers (Filler, decap, etc)

armleo, 2022-08-14:
In some lefs the CLASS is wrong. They need to be core spacer. Otherwise they are used in density calculations by OpenROAD.

Locations:

https://github.com/RTimothyEdwards/open_pdks/blob/6003ab31478184bceedc76eb650d8812e0cd6f5b/sky130/custom/sky130_fd_sc_hd/lef/sky130_ef_sc_hd__decap_12.lef#L6

https://github.com/RTimothyEdwards/open_pdks/blob/6003ab31478184bceedc76eb650d8812e0cd6f5b/sky130/custom/sky130_fd_sc_hd/lef/sky130_ef_sc_hd__fill_12.lef#L6

And in the sky130_fd_sc_hd_decap_* cells.,

Needs to be:
```
CLASS CORE SPACER ;
```

armleo, 2022-08-28:
I tried creating the patch files/pull requests, but because the test build required Magic VLSI + Open PDKs building, I am going to stop working on this issue.

Still issue needs fixing.

RTimothyEdwards, 2022-08-28:
This is easy enough to do in open_pdks;  I just didn't notice your issue posted.  I saw your update post, so I have gone ahead and fixed all the issues and am testing the build and will push when confirmed.

armleo, 2022-08-28:
@RTimothyEdwards Did you make the changes in the PDK cells too? (sky130_fd_sc_hd_decap_*)

RTimothyEdwards, 2022-08-28:
Yes, I did.

closed 2022-08-29 by armleo