fossi-foundation/open-pdks #274: GF180MCU lef and tlef metal layers name mismatch and missing SITE

kareefardi, 2022-08-31:
Not quite sure that this repo is the right place for this issue. `gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef` has uppercase metal layer names while `gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/techlef/gf180mcu_fd_sc_mcu7t5v0.tlef` has Pascal case metal layer names (`METAL1` vs `Metal1`). Furthermore, there are no `SITE` definitions in the techlef - nor the cells lef. Attached are build logs.
[build_gf180mcu-2022-08-31-04-33-01.tar.gz](https://github.com/RTimothyEdwards/open_pdks/files/9461640/build_gf180mcu-2022-08-31-04-33-01.tar.gz)

RTimothyEdwards, 2022-08-31:
This is an error in the Google repository and should be logged as an issue at either https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0 or at https://github.com/google/globalfoundries-pdk-libs-gf180mcu.  I expect Tim and Proppy would prefer if you just fix the technology LEF files and generate a pull request.

Note that this error also applies to the mcu9t5v0 standard cell library.

mithro, 2022-08-31:
Please **start** with logging the issue.

The fastest solution will be for you to send a fix pull request but maybe someone from MABrains can help fix it too? @atorkmabrains

kareefardi, 2022-09-01:
I logged both issues and created PRs for the metal layer names miss-match issue. I am working through the CLA for the PRs.

kareefardi, 2022-09-01:
I have signed an individual CLA. I think someone needs to re-trigger the CI.

atorkmabrains, 2022-09-02:
@kareefardi How can we help you?

mithro, 2022-09-05:
@atorkmabrains - The pull requests are at;
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/15
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/pull/17

kareefardi, 2022-09-08:
Fixed in google's repos

closed 2022-09-08 by kareefardi