fossi-foundation/open-pdks #279: ngspice lib files are not installed

satvikgoel950, 2022-09-08:
I installed the open_pdk using the procedures described but in the installation folder like /lib.tech folder in missing ngspice. Please look into it.
[image]

armleo, 2022-09-08:
Steps to reproduce?

armleo, 2022-09-08:
What steps did you use to install the PDK. Need full list of commands in correct order.

My suspect right now is that you ran out of space that's why I want the output of commands `df` and `quota -s`

satvikgoel950, 2022-09-09:
this is the list of commands after cloning the open_pdk from git
./configure --enable-sky130-pdk
make
sudo make install

here is the ouput of the the commands
1. df command
Filesystem     1K-blocks     Used Available Use% Mounted on
tmpfs             303508     1816    301692   1% /run
/dev/sda3       50770432 34175784  13983244  71% /
tmpfs            1517528        0   1517528   0% /dev/shm
tmpfs               5120        4      5116   1% /run/lock
/dev/sda2         524252     5364    518888   2% /boot/efi
tmpfs             303504     4708    298796   2% /run/user/1000

2. quota -s
no output(previously it didnt recognised the command then i installed quota using apt-get and then no output)

armleo, 2022-09-09:
1. post the magic build log
2. post magic building steps
2. post the pdk build log

satvikgoel950, 2022-09-09:
i didnt build and install magic myself it was part of a automated install of many tools
here is the git : https://github.com/kunalg123/vsdflow
should i just reinstall magic?

satvikgoel950, 2022-09-09:
[make.txt](https://github.com/RTimothyEdwards/open_pdks/files/9533912/make.txt)
here the log file of pdk

satvikgoel950, 2022-09-09:
[make.log](https://github.com/RTimothyEdwards/open_pdks/files/9533990/make.log)
here is the log file for magic

RTimothyEdwards, 2022-09-10:
From the make.log file:  `make[1]: Entering directory '/home/satvik/vsdflow/magic/magic-8.1.220'`

Magic version 8.1.220 is four years out of date and is not compatible with the open PDKs.

satvikgoel950, 2022-09-12:
i have reinstalled magic after cloning it from "https://github.com/RTimothyEdwards/magic" and have again tried to install open_pdk after running ./configure --enable-sky130-pdk command the result was this
[image]
the result shows "configure: Tools enabled for PDK setup installation: klayout magic netgen irsim openlane qflow xschem
" this also is missing ngspice please help

ChrisZonghaoLi, 2022-10-03:
To enable ngspice to work with sky130 under open_pdk environment, watch this tutorial video: https://www.youtube.com/watch?v=jXmmxO8WG8s