fossi-foundation/open-pdks #280: ngspice: sky130_fd_pr folder missing in sky130/sky130A/libs.ref/spi

nataraj-pinakapani, 2022-09-09:
I executed the following commands to install SKY13nm PDK. 

git clone git://opencircuitdesign.com/open_pdks
git clone https://github.com/google/skywater-pdk
cd skywater-pdk
git submodule init libraries/sky130_fd_io/latest
git submodule init libraries/sky130_fd_pr/latest
git submodule init libraries/sky130_fd_sc_hd/latest
git submodule init libraries/sky130_fd_sc_hvl/latest
git submodule update
cd ../open_pdks/
./configure --enable-sky130-pdk=/home/nataraj/projects/designmyic/cad/pdk_new/skywater-pdk --prefix=/home/nataraj/projects/designmyic/cad/pdk_new/sky_130nm --with-ef-style >  configure.log
make >  make.log
make install >  make_install.log

I got few errors while executing 'make' and 'make install. The errors are in make.log and make_install.log respectively.

After installation 'sky130_fd_pr' folder missing in sky130/sky130A/libs.ref/spi 
Files in this folder are referenced by ngspice models. 

Does any else have the same issue or can anyone please help me fix this issue?
[make.log](https://github.com/RTimothyEdwards/open_pdks/files/9539160/make.log)
[make_install.log](https://github.com/RTimothyEdwards/open_pdks/files/9539162/make_install.log)

RTimothyEdwards, 2022-09-09:
Seems likely that this is a duplicate of #279.  I'm taking a look into it.

nataraj-pinakapani, 2022-09-09:
Thank you very much

RTimothyEdwards, 2022-09-10:
`Magic 8.2 revision 157 - Compiled on Thu, 07 Nov 2019 21:14:45 +0100.`

This is your problem.  The most recent version of magic is `8.3.323`.  Your version is years old and not at all compatible with the SkyWater process.  There should be an error message about that in the output, although I think there is currently a bug in open_pdks such that the error doesn't stop the build process.

nataraj-pinakapani, 2022-09-17:
Thank you very much. That resolved the issue