fossi-foundation/open-pdks #287: Need magic to get open_pdks installed where magic was installed already

ChrisZonghaoLi, 2022-09-21:
So after I installed the magic (installed in my local directory since I do not have the server's root), I try to get back to install open_pdks, and it seems it cannot really detect the local directory where I installed my magic. Please see the picture below.

[image]

My magic executable is located here:
[image]

If you see the program, it does not complain about "klayout" since it was installed in the usr/ in the server, done by the server administrator long time ago and I think open_pdks has found it and therefore not complaining about it. But it stops at magic.

Being honest, is there a falg where I can customize which tools to install? In fact I am only looking to install xchem here (I do not see ngspice option so I assume I need to do this myself seperately).

Any idea on this? Thanks.

RTimothyEdwards, 2022-09-28:
Make sure that the path to magic's executable is in your PATH environment variable.  i.e.,
```
export PATH=${PATH}:~/cad/bin
```

closed 2022-10-03 by ChrisZonghaoLi

RTimothyEdwards, 2022-10-04:
I missed your second question.  If you do `./configure --help` in open_pdks, you'll see all the available options.  You can do, e.g., `--disable-magic`, `--disable-openlane`, etc., to avoid installing setup files for the tools that you don't need.  This is probably more trouble than it's worth, though, because the tool setups all have pretty low overhead;  collectively they use a tiny part of the disk space and take a tiny part of the build and install time.  It is also possible to do a more selective build of the PDK, but that is a bit more complicated.  You need to run `make` from the foundry directory (e.g., `sky130`) with very specific targets, such as `make xschem-A` for xschem and `make primitive-build-A` for the primitive device models and libraries.  I'm not sure how much that's worth the trouble, either, but it will save a huge amount of disk space, if that's valuable to you (you can also just delete the libraries you don't need from the `libs.ref` directory.  Make sure you keep `libs.ref/sky130_fd_pr`, which is the primitive device library and contains all the low-level component models, GDS, etc.).