fossi-foundation/open-pdks #289: sky130 diode parameter matching in LVS

d-m-bailey, 2022-09-28:
magic extracts `area` and `pj` for diodes in all variants except `orig`. 

xschem models only netlist `area`; thus, `pj` is flagged as a missing parameter when using xschem generated schematics.
The xschem models currently display no parameters in the schematic.

netgen reduces parallel diodes by adding `area`, but ignores `pj` during reduction.
`pj` should probably also be summed during parallel reduction.

I think a user friendly solution might be to use `w` and `l` as xschem parameters and use those to calculate `area` and `pj` when netlisting.

RTimothyEdwards, 2022-09-28:
@d-m-bailey :  I just fixed this in version 1.0.339.
The lack of perimeter value output in xschem will need to be taken up with Stefan Schippers.

StefanSchippers, 2022-09-28:
I can modify the diode symbol to require W and L parameters and output AREA and PJ in netlist,  however this change will probably break existing designs where diodes are placed with an AREA parameter.

Another option is to just add a perimeter parameter (PJ). If not present it can be estimated as 4* sqrt(AREA), otherwise it will be taken from instance parameter. Netlist will contain AREA=... and PJ=.. for diodes. This change will not beeak existing designs.

StefanSchippers, 2022-09-28:
OK will do asap

RTimothyEdwards, 2022-09-28:
No particular urgency.  It has a fairly easy workaround to edit the netgen setup script to ignore the `pj` parameter.

RTimothyEdwards, 2022-09-28:
That seems like an implementation error in ngspice.  I would expect either "1.0e-12" if ngspice was ignoring the SCALE option for diode area, or "1" if it was, but where in the world does "1.0e+12" come from??

StefanSchippers, 2022-09-28:
I think it's a scaling issue in the model. 
simulating a diode with AREA=1e12 and PJ=4e6 yields an I-V forward characteristic that makes sense.
[image]

RTimothyEdwards, 2022-09-28:
I think this is a diode level 3 issue, and it has to do specifically with how level 3 interprets "area" and "pj", which are _not_ in SI units:

```
The parameters CJSW and JSW are scaled by the instance parameter PJ whose default value is 0.0.

If L and W instance parameters are supplied, the diode is scaled by the factor: M*(L*SHRINK-XW)*(W*SHRINK-XW) otherwise it is scaled by M*AREA. 
```

RTimothyEdwards, 2022-09-28:
I pulled that off of some random web page of documentation.  It is not clear what is meant by "the diode is scaled by".

It's also not clear to me what happens if you pass L and W to the model.  According to the level 3 model description, the model should accept L and W as alternatives to AREA and PJ.  The level 3 diode is specifically supposed to be a semiconductor diode that scales according to physical dimension.  I am not sure of the status (or the correctness of the implementation) of level 3 diodes in either ngspice or Xyce.

StefanSchippers, 2022-09-28:
@RTimothyEdwards I have done some tests.
With ngspice i can specify W=.. L=... instead of AREA=... PJ=... 
 Putting W=1e6 L=1e6 i get same results as shown above for AREA=1e12 PJ=4e6.
 Again putting W=1, L=1 or even worse, W=1e-6 L=1e-6 creates a sub-atomic scaled diode.
 
 I think we need to decide how to proceed. 
 My desiderata is that a 1um x 1um silicon diode should be instantiated in a netlist as:
 dxxx n1 n2 diode_model AREA=1 PJ=4
 or equivalently:
 dxxx n1 n2 diode_model W=1 L=1

This is probably what matches LVS extracted diodes, but with current models this diode can not be simulated without rescaling dimensions.
I also tried with Xyce with no luck. models are not recognized at all and Xyce complains about missing models.

RTimothyEdwards, 2022-09-29:
You can ask Eric Keiter about level 3 diodes in Xyce.  I discussed it with him some time ago and he said he was open to implementing some basic support for level 3 diodes, which mainly means accepting the dimensional parameters.  But I have not heard of anything happening on that front since the discussion.

I would also prefer that the area and perimeter follow the syntax of all the other components in the process and use the same units.  That has always been the assumption of the magic extraction decks, and so the diodes in the extracted netlists (from, for example, the antenna diode standard cell) are already in these (incorrect) units.  I assume that the model equations can be rewritten to make the units work out correctly.

d-m-bailey, 2022-10-05:
@RTimothyEdwards So is the current consensus to ignore the `pj` parameter for diodes in LVS? There was a relatively recent change to magic that changed the extracted value. This caused all previously extracted diode parameters in the pdk to no longer match. However, if we're just going to ignore the value in the netgen setup file, it won't be an issue.

StefanSchippers, 2022-10-05:
Upstream [xschem_sky130](https://github.com/StefanSchippers/xschem_sky130) symbols for diodes will output Area and Pj. 
@RTimothyEdwards i also noticed that although a dc simulation of the diodes will show a "reasonable" I/V characteristic if using the unclear values Area=1e12 Pj=4e6 for something that looks like a 1um x 1um device, tran simulations do not work and diodes behave as open circuits (in Ngspice, Xyce does not process these device models at all). So my conclusion is that these models need a substantial fix. This also means that a reverse connected diode inserted by mistake somewhere will be unnoticed in simulation. Not ideal.
[image]

RTimothyEdwards, 2022-10-10:
@StefanSchippers :  Sorry, I have my head down working on the Caravel chip redesign.  Unless I rework the device models myself, it is probably the case that area for the diodes will have to be in the weird units of, I guess, picometers^2.  The PJ parameter does not appear in the device model, and I'm not sure if it is used or not (?).  The simplest solution is to keep the SkyWater definition, which would require me to modify the extraction definition of the diode in the tech file for magic to multiply the area value by 1E12 and the perimeter value (if used) by 1E6.