fossi-foundation/open-pdks #291: make: *** [Makefile:130: tech-sky130] Error 2

odevdeneme7896354125, 2022-10-04:
Hello, I'm new here. But I'm trying about 8 times but it didnt work. Every time make command runs about 1 hour but gives this error. Where can i look for it ?

# Remove the base verilog files which have already been included into
# the libraries
rm -f /home/cip/open_pdks/sky130/sky130A/libs.ref/sky130_fd_sc_lp/verilog/*.*.v
make[3]: Leaving directory '/home/cip/open_pdks/sky130'
make[2]: Leaving directory '/home/cip/open_pdks/sky130'
echo "Ended sky130A PDK staging on "`date` >> sky130A_make.log
make[1]: Leaving directory '/home/cip/open_pdks/sky130'
make: *** [Makefile:130: tech-sky130] Error 2

Here the last command line output. How can i fix this ? 

Also i'm getting "Permission denied" and "Could not write file.  Cell not written." outputs too. Is that normal or i must look for solution for it ?

-Best regards

RTimothyEdwards, 2022-10-04:
The build tends to keep going after an error, partly because its multithreaded, so the actual error may be a ways back.  Check the log file `sky130A_make.log` for when the actual fatal error occurred.

odevdeneme7896354125, 2022-10-05:
Well i dont see any fatal error in log file. Could you check for me ? Thank you. 
[Uploading sky130A_make-old.log…]()

with sudo su but output newer changed 

[Uploading sky130A_make.log…]()

odevdeneme7896354125, 2022-10-05:
I cant see any fatal errors. Can you check too ? Thank you
[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/9715630/sky130A_make.log)

RTimothyEdwards, 2022-10-07:
I see a lot of errors like:
```
Error message output from magic script:
Could not write file.  Cell not written.
```
and
```
sky130_ef_sc_hd__fill_8.mag: Permission denied
sky130_ef_sc_hd__fill_12.mag: Permission denied
```
That suggests that you might not have write permission into `/home/cip/`.

clocktop, 2022-10-08:
I am the same errors on install as odevdeneme7896354125, I have looked through my log file and do not see the same error message you mentioned in the prior post, instead I am seeing errors such as 

> Error message output from magic script:
Can't write file named '(UNNAMED)'

any tips on how I might proceed forward?

here is my log file for reference 
[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/9740021/sky130A_make.log)

RTimothyEdwards, 2022-10-08:
I don't see any final error in your `make` output, and I conclude that the build worked correctly.

RTimothyEdwards, 2022-10-08:
@clocktop :  I don't see any final error in your make output, and I conclude that the build worked correctly.

odevdeneme7896354125, 2022-10-08:
I spend a lot of time but last time i try Kali with "sudo su" command and it installed. Also magic ready for work. I want to share VM image but how can i share this image ? Whatever I'm kinda happy and I can sleep now. Thank you for helping me @RTimothyEdwards . I was thinking like why someone gonna help me but you helped me and made my day. So now I want to share my VM image for helping others. 

-Best regards

closed 2022-10-10 by odevdeneme7896354125