fossi-foundation/open-pdks #309: GF180 512x8 SRAM Macro Verilog model has 1.8V timing parameters

shalan, 2022-11-22:
This might be the case for other SRAM macros. 5.0V timing parameters should be used instead as Caravel runs @ a single 5.0v supply.

RTimothyEdwards, 2022-11-24:
@marwaneltoukhy :  We probably need a fork of the SRAM library on efabless, too, so we can get a prompt fix for the SRAM verilog.

RTimothyEdwards, 2022-12-01:
My understanding of the problem so far:

GF180MCU open PDK modifications to gf180mcu_fd_ip_sram:

Verilog files contain specify block timing that is taken from the 1.8V operation, which is not the normal operating voltage for the SRAM.

Existing gf180mcu_fd_ip_sram__sram512x8m8wm1.v:

          specparam Tcyc = 55600 : 55600 : 55600;
          specparam Tckh = 25000 : 25000 : 25000;
          specparam Tckl = 25000 : 25000 : 25000;

gf180mcu_fd_ip_sram__sram512x8m8wm1.v:
New values should be:

          specparam Tcyc = 11890 : 11890 : 11890;
          specparam Tckh =  4387 :  4387 :  4387;
          specparam Tckl =  5724 :  5724 :  5724;

gf180mcu_fd_ip_sram__sram256x8m8wm1.v:
New values should be:

          specparam Tcyc = 11000 : 11000 : 11000;
          specparam Tckh =  5401 :  5401 :  5401;
          specparam Tckl =  4969 :  4969 :  4969;

gf180mcu_fd_ip_sram__sram128x8m8wm1.v:
New values should be:

          specparam Tcyc = 10548 : 10548 : 10548;
          specparam Tckh =  4517 :  4517 :  4517;
          specparam Tckl =  4531 :  4531 :  4531;

gf180mcu_fd_ip_sram__sram64x8m8wm1.v:
New values should be:

          specparam Tcyc = 10280 : 10280 : 10280;
          specparam Tckh =  3963 :  3963 :  3963;
          specparam Tckl =  4846 :  4846 :  4846;

RTimothyEdwards, 2022-12-01:
Of the other things in the verilog file, these appear to be basically placeholders:

```
  specparam tcs  = 5000 : 5000 : 5000;
  specparam tas  = 5000 : 5000 : 5000;
  specparam tds  = 5000 : 5000 : 5000;
  specparam tws  = 5000 : 5000 : 5000;
  specparam twis = 5000 : 5000 : 5000;

  specparam tch  = 10000 : 10000 : 10000;
  specparam tah  = 10000 : 10000 : 10000;
  specparam tdh  = 10000 : 10000 : 10000;
  specparam twh  = 10000 : 10000 : 10000;
  specparam twih = 10000 : 10000 : 10000;
```
and maybe
```
  specparam Tdly  = 100 : 100: 100;
```
but this one appears to be in the same range as Tcyc, Tckh, and Tckl, and probably needs adjustment, but to what?
```
  specparam ta   = 45000 : 45000 : 45000;
```

RTimothyEdwards, 2022-12-02:
FYI, the values I am substituting come from this document:
https://gf180mcu-pdk.readthedocs.io/en/latest/IPs/SRAM/gf180mcu_fd_ip_sram/cells/gf180mcu_fd_ip_sram__sram512x8m8wm1/gf180mcu_fd_ip_sram__sram512x8m8wm1.html

azwefabless, 2022-12-04:
Tim/Shalan

Can you provide a link to the actual PDK SRAM .lib and VCS timing mode being used.

azwefabless, 2022-12-05:
The .lib for 5V typical is at: 

[https://github.com/google/globalfoundries-pdk-ip-gf180mcu_fd_ip_sram/blob/main/cells/gf180mcu_fd_ip_sram__sram512x8m8wm1/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_5v00.lib]

The verilog file in question is:
https://github.com/google/globalfoundries-pdk-ip-gf180mcu_fd_ip_sram/blob/main/cells/gf180mcu_fd_ip_sram__sram512x8m8wm1/gf180mcu_fd_ip_sram__sram512x8m8wm1.v

And finally the Datasheet is at: 
https://gf180mcu-pdk.readthedocs.io/en/latest/IPs/SRAM/gf180mcu_fd_ip_sram/cells/gf180mcu_fd_ip_sram__sram512x8m8wm1/gf180mcu_fd_ip_sram__sram512x8m8wm1.html

Our concern is that the verilog specparams appear to be set close to the numbers for the 1.8V operation in the datasheet.
Thx,
Andy

azwefabless, 2022-12-06:
GF has provided .db compiled data for corners but this does not address the root problem that the timing specs are clearly very different. Many specs present in the .v file are not reflected equivalently with specs in the .libs or .db