fossi-foundation/open-pdks #318: Compilation error in sky130_fd_sc_hd.v

kauser-rl, 2022-12-02:
In `pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v` line number `49162`, there appears to be a syntax error on the line. See the code below. It has an invalid label (should have been a comment). My guess is that in the Caravel SoC netlist simulation flow we don't see this because we always run with `USE_POWER_PINS` defined whereas I'm running on a netlist without the power pins.

From:
``
endif SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V
``

To:
``
endif // SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V
``

vijayank88, 2022-12-02:
This issue already reported in OpenLane https://github.com/The-OpenROAD-Project/OpenLane/issues/1121

RTimothyEdwards, 2022-12-02:
Now fixed by patch script in open_pdks version 1.0.365.