fossi-foundation/open-pdks #326: gf180mcu resistor property names in the setup file don't match spice

d-m-bailey, 2022-12-12:
The resistor properties in the gf180mcu setup file `l` and `w` don't match the resistor properties in the spice files `r_length` and `r_width`.
```
#-------------------------------------------
# Resistors (except metal)
#-------------------------------------------

set devices {}
lappend devices nwell_1p5
...

foreach dev $devices {
    if {[lsearch $cells1 $dev] >= 0} {
	property "-circuit1 $dev" parallel enable
	property "-circuit1 $dev" serial enable
	permute "-circuit1 $dev" 1 2
	property "-circuit1 $dev" merge {l ser_critical} {w add_critical}
	property "-circuit1 $dev" tolerance {l 0.01} {w 0.01}
	# Ignore these properties
	property "-circuit2 $dev" delete par1 pm
    }
    if {[lsearch $cells2 $dev] >= 0} {
	property "-circuit2 $dev" parallel enable
	property "-circuit2 $dev" serial enable
	permute "-circuit2 $dev" 1 2
	property "-circuit2 $dev" merge {l ser_critical} {w add_critical}
	property "-circuit2 $dev" tolerance {l 0.01} {w 0.01}
	# Ignore these properties
	property "-circuit2 $dev" delete par1 pm
    }
}

#-------------------------------------------
# RM (metal) resistors
#-------------------------------------------

set devices {}
lappend devices rm1
...

foreach dev $devices {
    if {[lsearch $cells1 $dev] >= 0} {
	property "-circuit1 $dev" parallel enable
	permute "-circuit1 $dev" 1 2
	property "-circuit1 $dev" tolerance {l 0.01} {w 0.01}
	# Ignore these properties
	property "-circuit2 $dev" delete par1 pm
    }
    if {[lsearch $cells2 $dev] >= 0} {
	property "-circuit2 $dev" parallel enable
	permute "-circuit2 $dev" 1 2
	property "-circuit2 $dev" tolerance {l 0.01} {w 0.01}
	# Ignore these properties
	property "-circuit2 $dev" delete par1 pm
    }
}
```

And from the extracted layout
```
X2 a_974_132# a_974_132# w_0_12# pplus_u r_width=20u r_length=22.78u
...
X0 PLUS MINUS VSUBS ppolyf_u r_width=2.5u r_length=2.8u
```

d-m-bailey, 2022-12-13:
The netgen documentation says that `add_critical` and `ser_critical` have been deprecated.

RTimothyEdwards, 2022-12-19:
I fixed this last week, so I am closing this issue.  Thanks for the report.

closed 2022-12-19 by RTimothyEdwards