fossi-foundation/open-pdks #329: DRC reference for MIM errors is being replaced.

d-m-bailey, 2022-12-19:
Looks like the string `MIM` is begin treated as a macro with a definition of `1` in the final version of `magic/gf180cmuC.tech`.

Here's a comparison of the expected vs actual. `MIMTM` has been replace with `1TM`. 
```
   width *mimcap 5000 "MiM cap top plate width < %d (MIMTM.8a)"                                  |   width *mimcap 5000 "MiM cap top plate width < %d (1TM.8a)"                                   
   spacing *mimcap *mimcap 600 touching_ok \                                                     |   spacing *mimcap *mimcap 600 touching_ok \                                                    
          "MiM cap top plate spacing < %d (MIMTM.6)"                                             |          "MiM cap top plate spacing < %d (1TM.6)"                                              
   surround mimcc mimcap 390 absence_illegal \                                                   |   surround mimcc mimcap 390 absence_illegal \                                                  
          "MiM cap must surround MiM cap contact by %d (MIMTM.4-VT.4)"                           |          "MiM cap must surround MiM cap contact by %d (1TM.4-VT.4)"                            
   spacing pad *mimcap 50 touching_illegal "MiM cap cannot overlap pad (MIM1M.X)"                |   spacing pad *mimcap 50 touching_illegal "MiM cap cannot overlap pad (11M.X)"                 
   spacing mimcc mimcc 480 touching_ok "MiM cap contact spacing < %d (MIMTM.9 - VT.3)"           |   spacing mimcc mimcc 480 touching_ok "MiM cap contact spacing < %d (1TM.9 - VT.3)"            
                                                                                                 |                                                                                                
   spacing via3 *mimcap 50 touching_illegal \                                                    |   spacing via3 *mimcap 50 touching_illegal \                                                   
          "MiM cap cannot overlap via3 (MIMTM.10)"                                               |          "MiM cap cannot overlap via3 (1TM.10)"                                                
   spacing *mimcap  *m4,rm4 1200 touching_ok \                                                   |   spacing *mimcap  *m4,rm4 1200 touching_ok \                                                  
          "MiM cap to Metal4 spacing < %d (MIMTM.1)"                                             |          "MiM cap to Metal4 spacing < %d (1TM.1)"                                              
   spacing *mimcap via4/m5  390 touching_illegal \                                               |   spacing *mimcap via4/m5  390 touching_illegal \                                              
          "MiM cap spacing to via4 < %d (MIMTM.5)"                                               |          "MiM cap spacing to via4 < %d (1TM.5)"                                                
   surround *mimcap m4 600 absence_illegal \                                                     |   surround *mimcap m4 600 absence_illegal \                                                    
          "Bottom plate overlap of MiM cap < %d (MIMTM.3)"                                       |          "Bottom plate overlap of MiM cap < %d (1TM.3)"                                        
   width mimcapc/m5 280 "MiM Contact width < %d (VT.1+VT.3)"                                     |   width mimcapc/m5 280 "MiM Contact width < %d (VT.1+VT.3)"                                    
                                                                                                 |                                                                                                
   variants (full)                                                                               |   variants (full)                                                                              
   cifmaxwidth mim_bottom_plate_space 0 bend_illegal \                                           |   cifmaxwidth mim_bottom_plate_space 0 bend_illegal \                                          
          "MiM bottom plate to unrelated metal < 1.2um (MIMTM.1)"                                |          "MiM bottom plate to unrelated metal < 1.2um (1TM.1)"                                 
   cifmaxwidth mim_bot_cont_surround 0 bend_illegal \                                            |   cifmaxwidth mim_bot_cont_surround 0 bend_illegal \                                           
          "MiM bottom plate surrounds contact < 0.4um (MIMTM.2)"                                 |          "MiM bottom plate surrounds contact < 0.4um (1TM.2)"                                  
```

One solution might be to define `MIM` to be `MIM` instead of `1`.

d-m-bailey, 2022-12-19:
Looks like `METAL5S` might have the same problem.

```
  #--------------------------------------------------------------                                |  #--------------------------------------------------------------
  # VIA 3 - Requires METALS4, METALS5, or METALS6 Module                                         |  # VIA 3 - Requires METALS4, 1, or METALS6 Module                                              
  #--------------------------------------------------------------                                |  #--------------------------------------------------------------
                                                                                                 |                                                                                                
   width v3/m3 280 "Via3 width < %d (V3.1 + 2 * V3.4)"                                           |   width v3/m3 280 "Via3 width < %d (V3.1 + 2 * V3.4)"                                          
   spacing v3 v3 240 touching_ok "Via3 spacing < %d (V3.2a - 2 * V3.4)"                          |   spacing v3 v3 240 touching_ok "Via3 spacing < %d (V3.2a - 2 * V3.4)"
   surround v3/m3 *m3 50 30 directional \                                                        |   surround v3/m3 *m3 50 30 directional \ 
          "Metal3 overlap of Via3 < %d in one direction (V3.3i - V3.3)"                          |          "Metal3 overlap of Via3 < %d in one direction (V3.3i - V3.3)"
+ +-- 15 lines: exact_overlap v3/m4--------------------------------------------------------------|+ +-- 15 lines: exact_overlap v3/m4-------------------------------------------------------------
   widespacing allm4,obsm4 10000 allm4,obsm4  300 touching_ok \                                  |   widespacing allm4,obsm4 10000 allm4,obsm4  300 touching_ok \                                 
          "Metal4 > %c spacing to unrelated m4 < %d (M4.2b)"                                     |          "Metal4 > %c spacing to unrelated m4 < %d (M4.2b)"
                                                                                                 |  
  variants *                                                                                     |  variants *
                                                                                                 |  
  #------------------------------------------------------                                        |  #------------------------------------------------------
  # VIA 4 - Requires METALS5 Module                                                              |  # VIA 4 - Requires 1 Module                                                                   
  #------------------------------------------------------                                        |  #------------------------------------------------------                                       
                                                                                                 |   
   width v4/m4 280 "Via4 width < %d (V4.1 + 2 * V4.4)"                                           |   width v4/m4 280 "Via4 width < %d (V4.1 + 2 * V4.4)"
   spacing v4 v4 240 touching_ok "Via4 spacing < %d (V4.2a - 2 * V4.4)"                          |   spacing v4 v4 240 touching_ok "Via4 spacing < %d (V4.2a - 2 * V4.4)"
   exact_overlap v4/m5                                                                           |   exact_overlap v4/m5 
                                                                                                 |  
  #-----------------------------                                                                 |  #-----------------------------
  # METALS 5 - METALS5 Module                                                                    |  # METALS 5 - 1 Module                                                                         
```

RTimothyEdwards, 2022-12-19:
I had previously been doing complicated `#undef`s when I encountered these (fixed them all in sky130;  hadn't really gone through gf180mcu to check for any), but defining the value as itself might be a simpler, clever solution (if it doesn't try to do a recursive replacement, which it might).

RTimothyEdwards, 2022-12-19:
Yes, defining something as itself does bad things.  But it's easy to put `#undef`s in a few key places and fix the problem.

RTimothyEdwards, 2022-12-19:
Fixed in the last push to the repo on opencircuitdesign.com.