fossi-foundation/open-pdks #330: gf180mcu possible drc rule problem.

d-m-bailey, 2022-12-19:
`gf180mcuC.tech`
```
 extend pfet,pcap *pdiff 280 exclusive "Transistor length < %d (PL.1a)"
 extend nfet,pcap *ndiff 280 exclusive "Transistor length < %d (PL.1a)"   <- maybe should be nfet,ncap
 width pfet,pcap 300 angles "Bent Transistor length < %d (PL.7)"
 width nfet,pcap 300 angles "Bent Transistor length < %d (PL.7)"   <- maybe should be nfet,ncap

 extend mvpfet,mvpcap *mvpdiff 500 exclusive "MV Transistor length < %d (PL.1a)"
 extend mvnfet,mvpcap *mvndiff 600 exclusive "MV Transistor length < %d (PL.1a)"   <- maybe should be mvnfet,mvncap
 width mvpfet,mvpcap 700 angles "Bent MV Transistor length < %d (PL.7)"
 width mvnfet,mvpcap 700 angles "Bent MV Transistor length < %d (PL.7)"   <- maybe should be mvnfet,mvncap
```

Is this ok?
```
 edge4way pfet,pcap,mvnnfet,mvpfet,mvpcap *poly/a 330 \   <- the mvnnfet seems like it might not belong here
        ~(*nsd,*mvnsd)/a (*pdiff,*mvpdiff)/a 300 \
        "n-ohmic spacing to PMOS gate < %d (NP.4b + PP.4c)"
 edge4way nfet,ncap,nnfet,mvnfet,mvncap *poly/a 330 \
        ~(*psd,*mvpsd)/a (*ndiff,*mvndiff)/a 300 \
        "p-ohmic spacing to NMOS gate < %d (PP.4b + NP.4c)"
```

RTimothyEdwards, 2022-12-19:
All valid points, and I will fix them.  Thanks!

RTimothyEdwards, 2022-12-19:
Fixed in the last push to the repo on opencircuitdesign.com.