fossi-foundation/open-pdks #331: gf180mcu standard cell spice library has diodes with unlabeled parameters

d-m-bailey, 2022-12-19:
`gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu9t5v0/spice/gf180mcu_fd_sc_mcu9t5v0.spice` has undefined values
```
.SUBCKT gf180mcu_fd_sc_mcu9t5v0__antenna I VDD VNW VPW VSS
D0 VPW I diode_nd2ps_06v0 0.2034p 1.85u M=1
D1 I VNW diode_pd2nw_06v0 0.2034p 1.85u M=1
.ENDS
```
whereas the extracted layout has
```
.subckt gf180mcu_fd_sc_mcu7t5v0__antenna VSS I VDD VNW VPW
D0 VPW I diode_nd2ps_06v0 pj=1.86u area=0.2052p
D1 I VNW diode_pd2nw_06v0 pj=1.86u area=0.2052p
.ends
```

RTimothyEdwards, 2022-12-19:
There is a CDL-to-SPICE script that does a regexp replacement there that apparently I copied and pasted from a line where I was doing a text deletion, and didn't edit it. . .

RTimothyEdwards, 2022-12-19:
@d-m-bailey :  Do you have any idea if it's valid CDL syntax to have an area and perimeter of a diode without any parameter name in front of it?  Feels like an error to me.

RTimothyEdwards, 2022-12-19:
There is at least one other major error in the CDL, which is that the `endcap` cell is given four ports in the CDL.  But `endcap` is just a copy of `filltie`, and has only two ports (VDD and VSS).  So the existence of at least one other major error raises the probability that the diode issue is also just a straight-up error from GlobalFoundries.

RTimothyEdwards, 2022-12-19:
Fixed in the last push to the repo on opencircuitdesign.com.