fossi-foundation/open-pdks #332: incorrect path to klayout lyp file

proppy, 2022-12-21:
Currently open_pdks installs technology file in the current directory layout:
```
/home/proppy/.volare/sky130A/libs.tech/klayout
/home/proppy/.volare/sky130A/libs.tech/klayout/drc
/home/proppy/.volare/sky130A/libs.tech/klayout/scripts
/home/proppy/.volare/sky130A/libs.tech/klayout/tech
/home/proppy/.volare/sky130A/libs.tech/klayout/tech/sky130A.lyp
/home/proppy/.volare/sky130A/libs.tech/klayout/tech/sky130A.lyt
/home/proppy/.volare/sky130A/libs.tech/klayout/lvs
/home/proppy/.volare/sky130A/libs.tech/klayout/pymacros
/home/proppy/.volare/sky130A/libs.tech/klayout/macros
/home/proppy/.volare/sky130A/libs.tech/klayout/ruby
/home/proppy/.volare/sky130A/libs.tech/klayout/python
/home/proppy/.volare/sky130A/libs.tech/klayout/klayoutrc
```
which is slightly different from the one in https://github.com/mabrains/sky130_klayout_pdk

This causes reference to the `lyp` file not be resolved my initializing klayout with technology:
```
$ KLAYOUT_HOME=$PDK_ROOT/sky130A/libs.tech/klayout klayout -e
...
Warning: Initialization of layers failed: Unable to open file: /home/proppy/.volare/sky130A/libs.tech/klayout/tech/sky130/sky130A.lyp (errno=2)
```

One can fix the reference to the `lyt` file to drop the `sky130` directory:
https://github.com/mabrains/sky130_klayout_pdk/blob/main/sky130_tech/tech/sky130/sky130.lyt#L7-L9
in order to workaround this issue:
```
 <base-path>$(appdata_path)/tech</base-path>
 <original-base-path>$(appdata_path)/tech</original-base-path>
 <layer-properties_file>sky130A.lyp</layer-properties_file>
```

xobs, 2023-01-10:
Which one is correct? sky130 and gf180 are definitely different, as evidenced by the fact that klayout doesn't work with gf180: https://github.com/The-OpenROAD-Project/OpenLane/issues/1593

proppy, 2023-07-12:
another error related to this is klayout failing to load the `lyp` file:
```
Warning: Initialization of layers failed: Unable to open file: /home/proppy/mixed-signal-sky130a-env/share/pdk/sky130A/libs.tech/klayout/tech//sky130.lyp (errno=2)
```

I think this is due to the installation renaming the file from being named after the `TECH` (`sky130`) to being named after the pdk variant `sky130A`:
https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/Makefile.in#L1011