fossi-foundation/open-pdks #333: scripts/update.sh missing from repo

antonblanchard, 2022-12-28:
The Makefiles refer to `scripts/update.sh` but I can't find it anywhere in the repo history.

RTimothyEdwards, 2022-12-29:
Good question.  The intent of the script was to do a 'git pull' if the library was a git repository, or to remove it and re-do the download process if not.  I'm surprised that it just isn't anywhere to be found, but it's easy enough to write the script and add it.

RTimothyEdwards, 2023-01-05:
Fixed in version 1.0.381.  The file was missing because there is a `.gitignore` file in the `scripts/` directory that is written to exclude all followed by a list of exceptions.  With `update.sh` never being added to that list, it never got committed to the repository.

antonblanchard, 2023-01-15:
This is now upstream, so closing

closed 2023-01-15 by antonblanchard