fossi-foundation/open-pdks #334: Line Ending Problem Unix/DOS

hakan-demirli, 2023-01-05:
I encountered the error first while I was building OpenRAM. If I run `make pdk` command in [OpenRAM repo](https://github.com/VLSIDA/OpenRAM) following error shows up.
```
...
make SHARED_PDKS_PATH=/pdk install"
sh: ./configure: /bin/bash^M: bad interpreter: No such file or directory
```
A WIndows-style line ending https://github.com/RTimothyEdwards/open_pdks/blob/d6c91b48a57fbe36eab06905211412327b0406dc/scripts/configure#L1  causes it. I can bypass it by running `dos2unix` over the `open_pdks` repo. But, then I have to follow make file instructions manually since specific commits are pulled and verified, which conflicts with the recent changes (dos2unix).

This is not a critical bug. Since the solution is obvious. But, I have encountered it on both Ubuntu 20.04 LTS, Ubuntu Server 22.04 LTS, and Ubuntu on WSL2. So, fixing it would be a mild convenience for the users.

RTimothyEdwards, 2023-01-05:
I have no idea what you're seeing.  The line you're pointing to does not contain a DOS-style ending and does not match the reported error (which is on /bin/bash, not /bin/sh).  I don't use Windows, so I'm mystified.

hakan-demirli, 2023-01-06:
**TLDR:** git automatically converts line endings if `autocrlf = true` is set in preferences. That was the case for me. I fixed it. All good.

> The line you're pointing to does not contain a DOS-style ending

This is what lead me to find the problem. Running `cat -e *` was showing Windows line endings (^M$). How could two people see different line endings? -> Git can change the line endings.
>  and does not match the reported error (which is on /bin/bash, not /bin/sh). I don't use Windows, so I'm mystified.

I have changed it to bash and deleted the space between `#!` and `/b...` just to try. I apologize for not reverting it back before opening an issue.

I have set the git preferences correctly. Everything works now.
```
[core]
	autocrlf = false
	eol = lf
```

closed 2023-01-06 by hakan-demirli

RTimothyEdwards, 2023-01-06:
Always good to have a record of obscure problems!