fossi-foundation/open-pdks #342: add gf180mcuD for 5LM + 11K top

jeffdi, 2023-01-18:


proppy, 2023-02-14:
assuming this is: https://gf180mcu-pdk.readthedocs.io/en/latest/analog/layout/inter_specs/inter_specs_3_43.html

RTimothyEdwards, 2023-02-14:
Completed with the merge of PR #351 today.

closed 2023-02-14 by RTimothyEdwards

proppy, 2023-02-15:
@RTimothyEdwards do we need also those to be updated?
- magic (tech file) for parasitics extraction
https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/magic/gf180mcu.tech
- OpenRCX for digital flow
https://github.com/RTimothyEdwards/open_pdks/tree/master/gf180mcu/openlane (were those provided by @tspyrou?)

/cc @msaligane

proppy, 2023-02-15:
@RTimothyEdwards should we keep this open until we get extraction and parasitics in?

msaligane, 2023-02-15:
I am still unclear if `metaltp` is the top metal layer or is it metal5?

proppy, 2023-02-15:
@msaligane according to my reading of https://gf180mcu-pdk.readthedocs.io/en/latest/analog/layout/inter_specs/inter_specs_3_43.html, TLM is metal5.

msaligane, 2023-02-15:
Okay, but then in the tech  LEF I see the other layers on top (6 etc..)
What is the top layer called METAL5 or METALTM?

RTimothyEdwards, 2023-02-16:
@msaligane :  In a very confusing nomenclature, GF declares that "METAL5" is the _design_ layer but "METALTM" is the _mask_ layer.  "METALTM" is only supposed to be used internally at GF.  However, the technology LEF files that I look at simply have Metal1 to Metal5 (for this stackup).  What technology LEF file(s) are you referring to?

RTimothyEdwards, 2023-02-16:
@proppy :  Yes, please keep this issue open until I have a chance to work through the parasitic tables for the 1.1um top metal.

msaligane, 2023-02-16:
Okay, thanks! I am referring to this one: https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/magic/gf180mcu.tech

msaligane, 2023-02-16:
I am trying to compare all the variants using that file since you use it to generate the tech LEF (I don't think that was a good idea). I am trying to see how much resistivity changes when you change from 9K to 11K.

RTimothyEdwards, 2023-02-16:
I don't use that file to generate the technology LEF files.  It is only for creating magic tech files.

RTimothyEdwards, 2023-02-16:
@msaligane :  Specifically, the technology LEF is handled in the Makefile by selecting the file that is appropriate for the stack, which was done manually by just defining the part of the technology LEF name corresponding to the PDK variant name A, B, C, or D:
```
# These definitions are used by the build recipes for the tech LEF
A_FULLSTACK = 3LM_1TM_30K
B_FULLSTACK = 4LM_1TM_11K
C_FULLSTACK = 5LM_1TM_9K
D_FULLSTACK = 5LM_1TM_11K
```

msaligane, 2023-02-16:
Thank you @RTimothyEdwards !

proppy, 2023-02-16:
> 
@proppy : Yes, please keep this issue open until I have a chance to work through the parasitic tables for the 1.1um top metal.

filed https://github.com/RTimothyEdwards/open_pdks/issues/352 and https://github.com/RTimothyEdwards/open_pdks/issues/353