fossi-foundation/open-pdks #343: SKY130 resistors do not cause thermal noise in ngspice

hpretl, 2023-01-26:
See https://sourceforge.net/p/ngspice/bugs/614

By default, `ngspice` **does not simulate thermal noise for behavioral resistors**, which are used in the SKY130 model files. Thermal noise can be enabled by adding the parameter `noisy=1` to the respective resistors.

As proper thermal noise modelling of resistors is essential for analog/mixed-signal circuit design, please patch the model files accordingly.

RTimothyEdwards, 2023-01-26:
Which resistor models does this affect?

hpretl, 2023-01-26:
At least `res_high_po`, `res_high_po_[0p35|0.69|1p41|2p85|5p73]`, `res_xhigh_po`, `res_xhigh_po_[0p35|0p69|1p41|2p85|5p73]`. Maybe more.

hpretl, 2023-02-10:
I fixed this on my end, see https://github.com/iic-jku/iic-osic-tools/commit/beaaa438b7d4b58b165106e8153fd94edb20187b.

It would still be good to add the `noisy=1` flag to behavioural resistors in the `SKY130` model files.