fossi-foundation/open-pdks #346: incorrect path to gf180mcu tech files for interactive usage

proppy, 2023-02-02:
in order to enabling interactive usage klayout expect the following directory hierarchy per: 
https://www.klayout.de/doc/about/technology_manager.html#:~:text=The%20technology%20folder%20may%20have%20subfolders%20to%20hold%20library%20files%2C%20macros%2C%20DRC%20runsets%2C%20LEF%20files%20and%20other%20pieces%20of%20the%20technology%20package

> Except for the default technology, technologies are kept in technology folders in KLayout's application path. They are read from subfolders from the "tech" directories. The technology definition itself is held in a file with extension ".lyt". The technology folder may have subfolders to hold library files, macros, DRC runsets, LEF files and other pieces of the technology package.

```
tech/
tech/pdkname.lyt
tech/pdkname.lyp
tech/pymacros/pdkname.lym # for autoload
tech/pymacros/cells # as of current lym file
tech/drc
tech/lvs
```

However [https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/Makefile.in and 
](https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/Makefile.in#L1008-L1012) and https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/Makefile.in#L660-L669 currently stage the file differently:
```
tech/
tech/pdkname.lyt
tech/pdkname.lyp
tech/pdkname.lym # doesn't autoload
pymacros/cells # not relative to lym
drc # override default base tech
lvs # override default base tech
```

It would be to re-organize those to follow klayout convention without breaking OpenLane and other tools expectation about the current layout (maybe a transition release with symlinks would be nice?)

RTimothyEdwards, 2023-02-02:
@kareefardi :  Can you please check how making this structural change to the directories in klayout affects the running of openlane?

proppy, 2023-02-15:
@RTimothyEdwards looking at the OpenLane code it seems that it does relies exclusively on `KLAYOUT_TECH` to resolve the tech files from klayout (assuming tcl forward its `$::env` to python invocation when not setting it explicitly):
https://github.com/The-OpenROAD-Project/OpenLane/blob/ce916510de89a037889e06ab16c00c4101f77d2f/scripts/tcl_commands/klayout.tcl#L38
https://github.com/The-OpenROAD-Project/OpenLane/blob/09e2095f9335981ec02c58b2f4d06f089448ed9e/scripts/klayout/open_design.py#L37-L38
https://github.com/The-OpenROAD-Project/OpenLane/blob/d8e399d55dfaac13048c84987ecf71f2086c5175/scripts/klayout/screenshot_layout.py#L36-L37
https://github.com/The-OpenROAD-Project/OpenLane/blob/ce916510de89a037889e06ab16c00c4101f77d2f/scripts/klayout/stream_out.py#L74-L75