fossi-foundation/open-pdks #350: Error regarding simulating inverter circuit in xschem with sky130

VanshikaTanwar, 2023-02-12:
While simulating the inverter circuit with sky130 model then at that time it's showing this error that library file not included not found while I have given the correct path for the library of sky130. And I have checked that tt file is present at that library location .
As , I am trying to simulate the inverter circuit which I have made using MOSFET using sky130 models NMOS and PMOS  then ,due to this error it's not generating the waveform of inverter circuit . 

Ngspice version - version= ngspice-36

I have attached the generated netlist below :-
** sch_path:
*+ /home/vanshikatanwar/Desktop/vsdflow/work/tools/xschem_git/xschem_simulations/inverter.sch
**.subckt inverter
Vin Vin GND 0
.save i(vin)
Vdd VDD GND 1.8
.save i(vdd)
XM1 Vout Vin GND GND sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM2 Vout Vin VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
**** begin user architecture code

.control
.dc Vin 0 1.8 0.01
.save all
.endc

**** end user architecture code
**.ends
.GLOBAL VDD
.GLOBAL GND
.end

[image]

[image]

[image]

RTimothyEdwards, 2023-02-12:
The specific error seems to be that it cannot find (path relative to the .lib file) `/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_pr/spice/sky130_fd_pr__nfet_01v8__tt.corner.spice`.  Do you not have that file?

VanshikaTanwar, 2023-02-13:
Yes, that file is present but ngspice file is not in the folder under libs.ref due to which the error was occuring ,so, I again installed the open_pdk then my error got solved . 
Thank you Rtimothy Edwards for your quick response .

closed 2023-02-13 by VanshikaTanwar