fossi-foundation/open-pdks #352: add parasitics extraction tables for 1.1um top metal

proppy, 2023-02-16:
As discussed in https://github.com/RTimothyEdwards/open_pdks/issues/342#issuecomment-1432256031, we need the magic parasitic extraction to account for the 11K variant.

Looking at https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/magic/gf180mcu.tech#L3658-L3659 it seems that `0P9` and `1P1` are currently assigned to the same resistance.

@RTimothyEdwards is that the only place that needs "updating"?

We were looking at the techlef in the 7t library with @msaligane, and it seems that both the `RESISTANCE` and `CAPACITANCE` and express in `PERSQ` units:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/blob/43beb45e4d323a76239de436db2df6732e9a689b/tech/gf180mcu_5LM_1TM_9K_7t_tech.lef#L275-L305
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/blob/43beb45e4d323a76239de436db2df6732e9a689b/tech/gf180mcu_5LM_1TM_11K_7t_tech.lef#L274-L304
Should we assume that the tools take into account the `THICKNESS` on top of this reference value in their calculation, or does that also need to  be updated.

proppy, 2023-02-21:
@RTimothyEdwards @jeffdi can we get a rought list of whats needed here, so that other people in the community can weight in to help?

RTimothyEdwards, 2023-02-21:
@proppy :  We have the metal stackup drawing and dimensions from GF for all process stackups including the one used for gf180mcuD, and that includes the dimensions for high/low/nominal process corners.  I have plugged all that information into my https://github.com/RTimothyEdwards/capiche system, which is cranking through it now.  I am still debugging a few things but should have results and push the capiche update within a few days.

proppy, 2023-02-21:
@RTimothyEdwards thanks for the update, is that https://github.com/RTimothyEdwards/capiche/commit/01dc26a9fcef2853f9af2f2815f6e644ccffb341 (the information you had to plug in capiche?)

RTimothyEdwards, 2023-02-21:
@proppy :  I am very bothered by the fact that GF posts exactly the same resistance for both 0.9um and 1.1um thicknesses of the top metal.  They also round off their dielectric constants to the nearest integer.  It is, however, the best I have to go on at the moment.

In principle the capacitance per unit distance should change, and it's sloppiness on GF's part not to update it for the 1.1um thick metal.  However, given the lack of precision in the various inputs, I don't think I could get a better number.  There won't be much difference in capacitance between 0.9um and 1.1um thick metal, so the existing technology LEF file is probably okay to use.

RTimothyEdwards, 2023-02-21:
@proppy :  That's the commit ID of the most recently pushed code, but I just added the GF high and low process corners, and that's still local to my machine and has not been pushed yet.

proppy, 2023-02-21:
> I am very bothered by the fact that GF posts exactly the same resistance for both 0.9um and 1.1um thicknesses of the top metal

If those are in `PERSQ` unit, should we expect to apply a multiplier that account for the `THICKNESS` on top of it?

RTimothyEdwards, 2023-02-21:
In the SkyWater case, the technology LEF file numbers were just plain wrong and didn't match their own published data.  In the GF case, the technology LEF file numbers match the published data, which also states that the 0.9um and 1.1um thick metals have the same resistance.  I don't want to second-guess GF without measurements to back up any assumptions.

proppy, 2023-03-08:
Looks like this was corrected with https://github.com/RTimothyEdwards/open_pdks/commit/815863e7bf1848eb9e21ac1fba50fba0fcbde173#diff-1fc31ec5402d966e4e4f163c0bfe8eca8ea31833cf5b19a595eb57f5100f746e ?

closed 2023-03-08 by proppy