fossi-foundation/open-pdks #353: add openrcx enablement for gf180mcuD

proppy, 2023-02-16:
Similar to the `rules.openrcx.*` files in https://github.com/RTimothyEdwards/open_pdks/tree/master/gf180mcu/openlane but for the `gf180mcuD` variant.

The process is documented here:
https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/rcx/calibration

/cc @tspyrou

minghungumich, 2023-02-22:
@proppy Is this what you are looking at? This is OpenROAD-flow-scripts openrcx rule file. https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/gf180/openROAD/rcx/gf180mcu_1p5m_1tm_11k_sp_smim_OPTB_typ.rules

proppy, 2023-03-08:
Looks like this was corrected with https://github.com/RTimothyEdwards/open_pdks/commit/815863e7bf1848eb9e21ac1fba50fba0fcbde173?

closed 2023-03-08 by proppy

RTimothyEdwards, 2023-03-08:
For the C variant (0.9um top metal), I had some OpenRCX rule files that I think came from the OpenROAD team.  I'll take a look at the files cited above.

proppy, 2023-03-09:
Thanks for adding the OpenROAD team rules too in https://github.com/RTimothyEdwards/open_pdks/commit/63dbde99bff08e6cfc41973a323fc846b63e8cb2, it's great that the PDK can ship both.