fossi-foundation/open-pdks #355: Metal5 width DRC wrong gf180mcuC

thesourcerer8, 2023-02-27:
Metal 5 is top level metal at 9K/11KAngstrom thickness (0.9um), so the rule that needs to be applied is MT.1, not M5.1. 
The KLayout DRC checks are currently checking the width for 0.28um, magic is checking for 0.36um, but actually MT.1 needs 0.44um (440nm).

proppy, 2023-02-27:
> The KLayout DRC checks are currently checking the width for 0.28um,

Are you referring to this check?
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv/blob/main/klayout/drc/rule_decks/metal5.drc#L24-L26

/cc @atorkmabrains

proppy, 2023-02-27:
Looking at the current checks would there be an issue applying both rules?

i.e: if you pass MT.1 (min 0.44µm) you should also pass M5.1 (min 0.28um).

thesourcerer8, 2023-02-27:
Yes, I am referring to these checks. Metal5 is actually the top metal layer, so the MT.1 rule has to be applied to the metal5. If the layer stackup would have 6 metal layers, then metal5 would need the normal M5.1 check, and instead metal6 would need to apply the MT.1 rule.

atorkmabrains, 2023-02-27:
@thesourcerer8 `MT` rules apply to the last metal in the stack. If you are using `gf180mcuC` then the last metal is 5.

RTimothyEdwards, 2023-02-27:
@atorkmabrains :  But it appears that file `metal5.drc` has a conditional `if METAL_LEVEL == "5LM"` which would imply that in a 5 metal stack, where metal 5 is the top metal, this rule deck is being applied to metal 5, which is incorrect.

atorkmabrains, 2023-02-28:
@RTimothyEdwards Yes, Metal5 rules and MT rules both should apply on the last metal in the metal stack as we have agreed earlier in our conversation about the layer GDS numbering issue.

If you look at the following link:
https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_02.html

My understanding that all the tapeouts follow the **DESIGN ACTIVITIES** table and we use the same layer numbering and we don't change the layer number of the last layer based on the stack. 

As the table highlights for `5LM`
```
Metal1, Via1, Metal2, Via2, Metal3, Via3, Metal4, Via4, Metal5.
Metal 5 must comply with MetalTop rule.
```

If you want to change this behavior, I'm more than welcome. But I need your confirmation here.

atorkmabrains, 2023-02-28:
@thesourcerer8 BTW, both are applied, you will have MT.1 (with 0.44um value) as well as M5.1 violation for klayout for that shape. If not, then please send the test case for review. You need to pass both rules btw.

atorkmabrains, 2023-02-28:
@RTimothyEdwards Moving forward, please take DRC rule deck from: https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv 

It has been reviewed and cleaned again. 

I'm planning to remove DRC from 
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr

LVS will be migrated soon.

atorkmabrains, 2023-03-01:
According to the discussion with @RTimothyEdwards , we will disable the Mn rules for top metal and only use MT rules.
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv/issues/85

atorkmabrains, 2023-03-02:
@RTimothyEdwards and @thesourcerer8 Fixed in https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv/commit/d6de5d063969c4fb031f862d4a964afafd07ad41

atorkmabrains, 2023-03-02:
@RTimothyEdwards Please update open_pdks and close the ticket afterwords.

RTimothyEdwards, 2023-03-02:
@thesourcerer8 :  Also I think I failed to note that your original statement said that _both_ klayout and magic were wrong (and in different ways).  I just updated open_pdks and fixed the issue with magic, as well as updating the commit manifest for the PV library repo, so when the PDK is rebuilt from scratch, the metal 5 DRC rules should be correct both in magic and in klayout.

FYI, going forward with GF Open MPW 2, it will use `gf180mcuD` with 1.1um thick top metal 5.  This is due to feedback from GF that the 0.9um thick top metal 5 has delamination issues when the pads are wire-bonded.   The 1.1um thick top metal option was apparently made by GF specifically to address that problem.