fossi-foundation/open-pdks #360: Many errors while building open_pdk

emfhasan, 2023-03-31:
I have run make command for building the open_pdk packages, but I came across the following errors for many unknown ef libraries and labels, can you advise: 

Error while reading cell "sky130_ef_io__vdda_lvc_pad" (byte position 2547962): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vccd_hvc_pad" (byte position 2547962): Cell "sky130_fd_io__overlay_vccd_hvc" is used but not defined in this file.
Error while reading cell "sky130_ef_io__vccd_hvc_pad" (byte position 2547962): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vddio_lvc_pad" (byte position 2547962): Cell "sky130_fd_io__overlay_vddio_lvc" is used but not defined in this file.
Error while reading cell "sky130_ef_io__vddio_lvc_pad" (byte position 2547962): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): Cell "sky130_fd_io__top_gpiov2" is used but not defined in this file.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): Cell "sky130_fd_io__overlay_gpiov2" is used but not defined in this file.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vccd_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__top_power_hvc_wpadv2 was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__top_gpiov2 was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssa_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vddio_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssd_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vdda_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__corner_bus_overlay was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssa_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssio_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__top_power_lvc_wpad was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vddio_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__top_ground_lvc_wpad was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssd_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vdda_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vssio_hvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__top_ground_hvc_wpad was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_vccd_lvc was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 2547962): cell sky130_fd_io__overlay_gpiov2 was used but not defined.
Error while reading cell "sky130_fd_io__simple_pad_and_busses" (byte position 278344): Cell "sky130_fd_io__com_busses_esd" is used but not defined in this file.
Error while reading cell "sky130_fd_io__simple_pad_and_busses" (byte position 278344): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__analog_pad" (byte position 278344): cell sky130_fd_io__com_busses_esd was used but not defined.
Error while reading cell "sky130_ef_io__gpiov2_pad_wrapped" (byte position 23648): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3221094): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3221126): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223718): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223750): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223782): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223814): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223846): NODE elements not supported: skipping.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 3223942): NODE elements not supported: skipping.
CIF file read warning: Input off lambda grid by 4/5; snapped to grid.
Error while reading cell "sky130_fd_io__top_gpiov2" (byte position 11194436): Warning:  Cell sky130_fd_io__top_gpiov2 boundary was redefined.
Error while reading cell "sky130_fd_io__gpio_ovtv2_hotswap_pghspu" (byte position 39937486): Warning: cell "sky130_fd_pr__pfet_01v8__example_55959141808164" placed on top of itself.  Ignoring the extra one.
Couldn't find label AMUXBUS_A
Couldn't find label AMUXBUS_B
Couldn't find label VSSA
Couldn't find label VDDA
Couldn't find label VSWITCH
Couldn't find label VDDIO_Q
Couldn't find label VCCHIB
Couldn't find label VDDIO
Couldn't find label VCCD
Couldn't find label VSSIO
Couldn't find label VSSD
Couldn't find label VSSIO_Q
Couldn't find label VCCD_PAD
Can't write file named '(UNNAMED)'

d-m-bailey, 2023-03-31:
@emfhasan Most (if not all) of these "errors" can be considered to be warnings, I believe. Did `make open_pdks` finish?

RTimothyEdwards, 2023-03-31:
Yes, I have not done much clean-up of the open_pdks output, which runs a lot of different tasks and attempts to work around errors using different methods, but then dumps all of the error messages.  As long as the "make" process finishes without raising an error itself and stopping early, then it has properly built the PDK.