fossi-foundation/open-pdks #361: Running ngspice for sky130 process and open_pdks

emfhasan, 2023-04-06:
I am trying to use the ngspice simulator with xschem using skywater130 process and open_pdk. I got the following error when I was running one test example: 
Error: Could not find include file ~/home/eda_work/eman/hassan/skywater-pdk/libraries/sky130_fd_pr_ngspice/latest/corners/tt.spice

I followed the instruction in the following video for [Stefan Schippers](https://www.youtube.com/@stefsport2002) to install the open_pdk, xschem for sky130 process and ngspice: https://www.youtube.com/watch?v=jXmmxO8WG8s

Any advice in this regard?

RTimothyEdwards, 2023-04-06:
If you installed the PDK through open_pdks, then the include path you cited is incorrect;  it is a path to the PDK sources, not the installed PDK (also, `sky130_fd_pr_ngspice` is not a valid directory name in either case).

Also, you do not want to include the corner file directly, but specify the top level file of models using the `.lib` card along with the name of the corner.

If you installed the open_pdks in the default location, which is `/usr/local/share/pdk`, then the proper way to include the models would be

```
.lib /usr/local/share/pdk/sky130A/libs.tech/ngspice/sky130.lib.spice tt
```

emfhasan, 2023-04-07:
Thnak you for the response. However, I installed the pdk in my local diretory **not** the /usr/local/share/pdk. the attached path is the error msg i got when i tried to simulate some example. When i installed the xschem i use to modify the following lines in  xschemrc (as recommended in the[Stefan Schippers](https://www.youtube.com/@stefsport2002)):

 set SKYWATER_MODELS ~/home/eda_work/eman/hassan/skywater-pdk/libraries/sky130_fd_pr_ngspice/latest
 set SKYWATER_STDCELLS ~/home/eda_work/eman/hassan/skywater-pdk/libraries/sky130_fd_sc_hd/latest

However, the default path to Models and Stdcells are: 
 set SKYWATER_MODELS ${PDK_ROOT}/${PDK}/libs.tech/ngspice
 set SKYWATER_STDCELLS ${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_sc_hd/spice

RTimothyEdwards, 2023-04-07:
I do not recommend including files from the source paths.  The sources don't have a top level library for handling corners.  The file `sky130.lib.spice` is a custom file in open_pdks that handles all of the includes of all model files for every corner.

But I think more to the point:  As I said earlier, there _is no such library_ `sky130_fd_pr_ngspice` so if you try to include that it is always going to fail.

emfhasan, 2023-04-07:
Can you please clarify where I should add the .lib /usr/local/share/pdk/sky130A/libs.tech/ngspice/sky130.lib.spice tt command?

Also, I have noticed that I have two installations of xschem-sky130 for the SkyWater process. One was installed as per the instructions and is located in ~open_pdks/sources/xschem_sky130 (which seems to come with open_pdks), and the other was manually installed in my home directory. Would it be best to remove the manually installed one to avoid confusion in the future?

Thank you for being patient to my many, very basic questions!

RTimothyEdwards, 2023-04-07:
The `.lib` command replaces any `.include` commands at the top of the testbench netlist.  In xschem, it would be part of a "code" block in the schematic.

If you let open_pdks clone the repositories for you, then you don't need to have your own copy outside of open_pdks/sources/.  You can also pass the location of your repository to the "configure" script for open_pdks, which will prevent it from cloning its own copy.  Either method is fine, but you don't need two copies of the repository.