fossi-foundation/open-pdks #362: issue in xschem and Ngspice using sky130 process

emfhasan, 2023-04-07:
I have an issue when try to simulate the Reram-tb utilizing xscehm and ngspice. I encounter the following errors when trying to simulate the reram and check the waveform uisng ngspice: % l_s_d(): Symbol not found: /home/schippes/share/xschem/xschem_library/devices/vsource.sym
l_s_d(): Symbol not found: /home/schippes/share/xschem/xschem_library/devices/vsource.sym
Hoever, the library for the reram is installed with other sky130 libraries in my local directory not the path given in the error message.

when I press on the voltage source and edit the path for vsource to be my local directory for vsource I got the following error:

Circuit: ** sch_path: /home/eda_work/eman/hassan/open_pdks/sources/xschem_sky130/sky130_tests/tb_reram.sch

Error on line 3 or its substitute:
n1 top 0 sky130_fd_pr_reram__reram_model
unknown device type - error
The simulation interrupted due to an error!

Also, when cant get the Netlist for the design when I press the Create Netlist button?!.

I am totally new to this technology and I would appreciate your help.
Thank you.

RTimothyEdwards, 2023-04-07:
You should look at the discussions in the #reram channel of the open-source-silicon Slack workspace.

The line `n1 top 0 sky130_fd_pr_reram__reram_model` starts with "N" and indicates an OSDI/OpenVAF model.  See:  https://ngspice.sourceforge.io/osdi.html

Running ngspice on this model requires the most recent version of ngspice compiled with OSDI/OpenVAF support.

emfhasan, 2023-04-07:
Could you please share the link to the #reram channel of the open-source-silicon Slack workspace? I attempted to join the workspace at https://open-source-silicon.dev/, but my account could not be linked to the Slack channel. Any assistance you can provide would be greatly appreciated.

When I was reviewing the reram library, I noticed that it did not have a model folder like the other libraries available. Instead, there was an empty third-party/make-env folder. I am confused about why the simulator was initially searching for the missing vsource in the /home/schippes/share/xschem/xschem_library/devices/vsource.sym directory, despite the fact that I have it in my local directory at eman/hassan/xschem/xschem_library/devices/vsource.sym. I am unsure why this issue occurred, or may I missed something during installation!. 

As I mentioned, I am totally new to this technology and I am starting recently my learning stage in the design area. 
Thank you.

RTimothyEdwards, 2023-04-07:
The registration for the Slack workspace was https://join.skywater.tools but I think there is an expired certificate owned by Google which has not been renewed.  Does it give you a registration option if you just go directly to this page: https://open-source-silicon.slack.com/?

RTimothyEdwards, 2023-04-07:
This is the direct link:  https://join.slack.com/t/open-source-silicon/shared_invite/zt-18steaudi-vFYEnjabWFspgA4_iiHkMw