fossi-foundation/open-pdks #376: Error while installing skywater PDK during make install step

aakash-ics, 2023-05-21:
[image]

I am getting this error multiple times while installing skywater pdk

Kindly help.

Thanking you
Aakash Gupta

RTimothyEdwards, 2023-05-21:
This has been reported before as an error which is, I think, specific to Debian systems, and has something to do with the conda setup.

The prior recommendation was just to re-run "make" and the problem resolves itself.  But you say you're getting this error multiple times?

There has been a suggestion that the problem can be solved on Debian systems by first doing `conda update -y conda`.   Please give that a try and let me know if it solves the problem or not.

proppy, 2023-05-23:
@aakash-ics, can you share your environment.yml?

proppy, 2023-05-23:
after digging into this with @mithro, it appears that the https://anaconda.org/symbiflow channel doesn't have the yosys package anymore: https://anaconda.org/litex-hub/yosys is the channel most recent conda package for most tools.

I'm not sure why this happened, but this could easily be fixed by changing https://github.com/google/skywater-pdk/blob/main/environment.yml#L18 to point to `litex-hub`.

proppy, 2023-05-23:
Looks like the yosys package was removed by @PiotrZierhoffer 1+ year go:
[image]

So I'm not sure why this started surfacing now?

aakash-ics, 2023-05-28:
Hi Proppy, 

Your Solution worked(and I installed miniconda) when I did make timing. But I am facing a new problem. 
when we make the PDK after doing ./configure in openpdks directory. the make[3]leaving directory step is taking forever.
[image]
[image]

KamikazeVildsvin, 2023-05-30:
I currently also stuck at the "leaving directory" point in the make process. Screenshot below. I have been able to install other tools like `magic` and `xschem` from source.
[image]

RTimothyEdwards, 2023-05-30:
The "make" process is multi-threaded, so what is happening is that one thread finished but another one is still going.  There are a couple of processes that take a long time---one of them is generating the I/O libraries, and the other is generating the SRAM libraries.  The time to build can vary greatly depending on the speed and number of cores of your computer.  At most it might take a few hours to complete.  How long did you give it to finish?

aakash-ics, 2023-05-30:
Hi RTimothyEdwards,
the make process indeed took quite a lot of time in hours but the problem here is it kind of gets stuck at this step(I kind of gave this step an hour, this make[3]: leaving directory step) not moving forward or throwing an error. Its just stuck at this step make[3]:leaving directory step of the make step.

Thanks
Aakash Gupta

KamikazeVildsvin, 2023-05-30:
@RTimothyEdwards, indeed after letting the "make" process sit for about four or five hours, it worked. I guess it just taks a loooot of time with the SRAM part, as you said. Thanks for the help! @aakash-ics, how long did you wait? Did you get it to work?