fossi-foundation/open-pdks #381: Soft connection checking uses a modified tech file that must be updated when the base file changes.

d-m-bailey, 2023-06-15:
@RTimothyEdwards We've talked about implementing the no-well extraction technology as a variant in the magic tech file.

Here are the base version and the no well version of the tech files I'm using for sky130A.

Is this enough information to create a variant? (maybe `nowell`)

[tech.tgz](https://github.com/RTimothyEdwards/open_pdks/files/11755146/tech.tgz)

RTimothyEdwards, 2023-06-15:
The device terminals are a general problem in that they are always "soft" connections---there is always some distance between the point under the device and the nearest well/substrate tap.  This is generally enforced by DRC rules, although the DRC rules tend to be rather computationally expensive.  I think there should be a way for the connectivity function to create for itself a different connectivity table which excludes the well and substrate types, and be able to apply that table selectively.  That could be set based on a criterion for type resistivity, to exclude the high-resistance well and substrate types automatically.  I have to look at the definition of the resistance calculation (the one in "extract", not "extresist") to see if it is sufficient to make a "soft" threshold based on wire resistance, which could be used to find soft connections through long poly or local interconnect wires.

RTimothyEdwards, 2023-06-15:
@d-m-bailey :  This would be a sort of coarse-grained check, but I'd like to know for layouts that have soft connection errors and those that use poly or local interconnect as long resistive routes, whether they show up clearly in the `.ext` file in the `node` statements for the net in question.  The `node` line syntax starts out `node <name> <R> <C> ...` where `<R>` is a resistance calculated from the area and perimeter of each layer in the net.  It should catch soft errors, but I just don't know how reliably.  The calculation of `<R>` assumes a simple point-to-point wire, and anything with branches or loops or other complex geometry is going to have an inaccurate result.

d-m-bailey, 2023-06-16:
@RTimothyEdwards I think I have a solution using variants. I'll post it soon.

d-m-bailey, 2023-06-17:
Should be fixed by #382