fossi-foundation/open-pdks #387: Latest build of `sky130` pdk doesn't include sram macros

passant5, 2023-07-17:
Open PDKs commit `78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc` doesn't include `sky130_sram_macros`  
```
> ls volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/*/libs.ref
volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/sky130A/libs.ref:
sky130_fd_io  sky130_fd_pr  sky130_fd_sc_hd  sky130_fd_sc_hvl  sky130_ml_xx_hd

volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/sky130B/libs.ref:
sky130_fd_io  sky130_fd_pr  sky130_fd_sc_hd  sky130_fd_sc_hvl  sky130_ml_xx_hd
```
the previous commit `3df14f84ab167baf757134739bb1d2c5c044849c` includes it
```
> ls volare/sky130/versions/3df14f84ab167baf757134739bb1d2c5c044849c/*/libs.ref
volare/sky130/versions/3df14f84ab167baf757134739bb1d2c5c044849c/sky130A/libs.ref:
sky130_fd_io  sky130_fd_pr  sky130_fd_sc_hd  sky130_fd_sc_hvl  sky130_ml_xx_hd  sky130_sram_macros

volare/sky130/versions/3df14f84ab167baf757134739bb1d2c5c044849c/sky130B/libs.ref:
sky130_fd_io  sky130_fd_pr  sky130_fd_sc_hd  sky130_fd_sc_hvl  sky130_ml_xx_hd  sky130_sram_macros
```

RTimothyEdwards, 2023-07-17:
@passant5 :  This strikes me as a volare issue;  `sky130_sram_macros` is a configuration option that is disabled by default (because it takes so long to build and install, although I think that's no longer such an issue).  Is the volare configuration enabling it?

passant5, 2023-12-12:
It was a volare issue where it omitted `sky130_sram_macros` even though the configuration enabled it. Fixed in volare

closed 2023-12-12 by passant5