fossi-foundation/open-pdks #389: Error while running "make" in Ubuntu 22.04 WSL2

akmalns, 2023-08-01:
Hi Everyone, I tried to install open pdk in my Ubuntu 22.04 WSL2 system. While running the "make" command, I got the following error message:

`Modifying port order in sky130B/libs.ref/sky130_fd_pr/maglef/sky130_fd_pr__rf_pfet_01v8_bM04W5p00L0p25.mag
Error:  Cannot find file sky130B/libs.ref/sky130_fd_pr/mag/sky130_fd_pr__rf_pfet_01v8_mvt_aF02W0p84L0p15.mag
Adding special extraction property to device sky130_fd_pr__rf_pfet_01v8_mvt_aF02W0p84L0p15 layout
Altering gencell property in device sky130_fd_pr__rf_npn_05v5_W1p00L1p00 layout
Altering gencell property in device sky130_fd_pr__rf_npn_05v5_W1p00L1p00 layout
Altering gencell property in device sky130_fd_pr__rf_npn_05v5_W1p00L2p00 layout
Altering gencell property in device sky130_fd_pr__rf_npn_05v5_W1p00L2p00 layout
Altering gencell property in device sky130_fd_pr__rf_pnp_05v5_W0p68L0p68 layout
Error:  Cannot find file sky130B/libs.ref/sky130_fd_pr/maglef/sky130_fd_pr__rf_pnp_05v5_W0p68L0p68.mag
Altering gencell property in device sky130_fd_pr__rf_pnp_05v5_W3p40L3p40 layout
Error:  Cannot find file sky130B/libs.ref/sky130_fd_pr/maglef/sky130_fd_pr__rf_pnp_05v5_W3p40L3p40.mag
Altering gencell property in device sky130_fd_pr__rf_npn_11v0_W1p00L1p00 layout
Altering gencell property in device sky130_fd_pr__rf_npn_11v0_W1p00L1p00 layout
make[3]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make[2]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make[1]: *** [Makefile:1022: all-B] Error 2
make[1]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make: *** [Makefile:132: tech-sky130] Error 2`

I have tried running this for multiple time but I still get similar but not the same error like this:

`
puts stdout "Done."
quit -noprompt

---------------------------------------------------------------------------

make[3]: *** [Makefile:1665: digital-lp-B] Error 1
make[3]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make[2]: *** [Makefile:1403: digital-lp-build-B] Error 2
make[2]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make[1]: *** [Makefile:1022: all-B] Error 2
make[1]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make: *** [Makefile:132: tech-sky130] Error 2
`

I also tried to run it with sudo, however, the same error keep popping up.

Are there any solutions for this problem until now? Thank you in advance

RTimothyEdwards, 2023-08-01:
@akmalns :  Because the Makefile process is multithreaded, errors can happen a long time before the process finishes generating output, so you will probably have to look through the log file to find the original error.  Errors like "Cannot find file" are probably secondary.  However, the most probable cause of failure is that your version of Magic does not match the minimum version requirement for the tech file.  What is your version of magic, and what is the version of open_pdks?

akmalns, 2023-08-01:
Hi @RTimothyEdwards , Thank you for your reply. May I know how I can look into the log file? My Magic version is 8.3.406 and the open pdk version is 1.0.431

RTimothyEdwards, 2023-08-01:
You need magic version 8.3.411 or newer.

The terminal output is copied to a log file that is in the open_pdks/sky130/ directory, e.g., sky130A_make.log.

Scorpion-DX, 2023-08-02:
> You need magic version 8.3.411 or newer.
> 
> The terminal output is copied to a log file that is in the open_pdks/sky130/ directory, e.g., sky130A_make.log.

Hello Timothy, 

I met the same problem when trying to compile open pdk on my ubuntu 20 virtual machine.
I'm using magic 8.3 revision 50.

The logs on my side looks like this.  In these logs I see there are some version error in the techfile.
[image]

My logs are attached below.
[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/12238269/sky130A_make.log)
[sky130B_make.log](https://github.com/RTimothyEdwards/open_pdks/files/12238271/sky130B_make.log)

Could you give me some suggestions on this?

akmalns, 2023-08-02:
Hi Timothy,

I updated my magic version to 8.3.421. It run past the previous point. However, it did stop again with the following error.
```
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vssd_hvc
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vssio_hvc
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vdda_hvc
# Add a maskhint set for the GPIO pad .mag view to prevent problems writing
# when writing HVI to GDS during hierarchical adjustments.
../common/insert_property.py -std_format /home/akmalns/Downloads/open_pdks/sky130/sky130B sky130_fd_io sky130_fd_io__top_gpiov2 \
        "MASKHINTS_HVI 1346 17198 5828 19224 13700 1890 15920 2360 24 17522 1778 20612" -mag
insert_property.py:
   source   = /home/akmalns/Downloads/open_pdks/sky130/sky130B
   library  = sky130_fd_io
   cell     = sky130_fd_io__top_gpiov2
   property = MASKHINTS_HVI 1346 17198 5828 19224 13700 1890 15920 2360 24 17522 1778 20612
make[3]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make[2]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
echo "Ended sky130B PDK staging on "`date` >> sky130B_make.log
make[1]: Leaving directory '/home/akmalns/Downloads/open_pdks/sky130'
make: *** [Makefile:132: tech-sky130] Error 2
```

My logs are attached below.

[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/12239015/sky130A_make.log)
[sky130B_make.log](https://github.com/RTimothyEdwards/open_pdks/files/12239072/sky130B_make.log)

Thank you in advance

Scorpion-DX, 2023-08-02:
> 

Maybe have a try on magic 8.3.411, it worked for me

Scorpion-DX, 2023-08-02:
> Hi Timothy,
>
> I updated my magic version to 8.3.421. It run past the previous point. However, it did stop again with the following error.
>
> ```
> No CDL file contains sky130_fd_io device [...]

Maybe have a try on magic 8.3.411, it worked for me

RTimothyEdwards, 2023-08-02:
@akmalns :  The error in the log file suggests there was a non-ASCII character in a property line of a magic file.  I have no idea where that would be coming from.  It could happen if you had a non-ASCII character in a directory path.  The error is in sky130A_make.log at line 12876 (python UnicodeDecodeError).  You could try reading the file in iso8859-1 encoding, which would involve changing common/foundry_install.py line 346 to `with open(magfile, 'r', encoding='iso8859-1') as ifile:` which would probably get it through the PDK installation without throwing an error, but regardless I think there's an underlying error somewhere, and it's one I've never seen before.

akmalns, 2023-08-07:
@RTimothyEdwards @Scorpion-DX 

I tried to remove all the existing files and started from the beginning with the correct version of magic (8.3.411) and It succeeded. Therefore, I conclude that the problem should be caused by incorrect version of magic. After upgrading the magic version, we can try running "make" again. If the error persist, try to start from the beginning.

Thank you for all of the support

closed 2023-08-07 by akmalns