fossi-foundation/open-pdks #397: Efabless cell LEF files do not correctly classify `VNB`, `VPB` as power pins

donn, 2023-09-12:
Consider this part of the LEF for `sky130_ef_sc_hd__decap_12`:

https://github.com/RTimothyEdwards/open_pdks/blob/ebba698e38217b7af5817de5ff6e7b1f4be45fa1/sky130/custom/sky130_fd_sc_hd/lef/sky130_ef_sc_hd__decap_12.lef#L24-L36

Designs from the PDK proper lack those pins entirely, but when `open_pdks` "injects" the pins, they're injected as follows:

```lef
PIN VNB
    DIRECTION INOUT ;
    USE GROUND ;
    PORT
      LAYER pwell ;
        RECT 0.005 0.785 0.925 1.015 ;
        RECT 0.005 0.105 3.590 0.785 ;
        RECT 0.150 -0.085 0.320 0.105 ;
    END
  END VNB
  PIN VPB
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER nwell ;
        RECT -0.190 1.305 3.870 2.910 ;
    END
  END VPB
```

`USE POWER ;` and `USE GROUND ;` are used by OpenROAD to determine if the pins in question are power pins or not, so when writing an unpowered netlist, virtually every `sky130_ef_sc_hd__decap_12` is broken.

This looks like a straightforward fix. Should I submit a PR or is there a reason behind this?

donn, 2023-09-12:
This appears to have been a regression in https://github.com/RTimothyEdwards/open_pdks/commit/1341f54f5ce0c4955326297f235e4ace1eb6d419

RTimothyEdwards, 2023-09-12:
Probably an item that I missed when transferring the PDK from Google to Efabless and attempting to clean up all the files in open_pdks accordingly.  Yes, the right solution is just to add the lines to the LEF files in the custom/ directory.  Go ahead and submit a PR.

closed 2023-09-14 by donn