fossi-foundation/open-pdks #409: Symbol calculated resistance on fixed width high poly resistors does not appear correct

mhommelga, 2023-10-24:
There is a large discrepancy between the calculated resistance (from L & W parameters) on the high poly resistor symbols depending on whether the fixed width or custom width symbol is used.
Attached picture shows the issue and also shows that xhigh poly resistors match perfectly.

[image]

Example:
 Custom Width: Symbol: res_high_po W=1.41um L=10um Rcalc=2251 Ohm
 FIxed Width: Symbol res_high_po_1p41 (W=1.41um)  L=10um Rcalc=1632Ohm

In some cases the values differ by 5x and the custom width resistors appear to be more realistic values and better match the simulation model. However for LVS to work on the layout I have, I need to use the fixed width resistors in the schematics.

mhommelga, 2023-10-24:
Thanks - indeed the values shown on the symbol differs from the simulated value.
Still someone might trust the symbol value blindly and not double check the simulation.
As mentioned in my setup the only resistors I can get LVS clean are the fixed width ones. The others cannot be matched by netgen.

It would just be good to have the calculated resistance value match regardless of which version of the resistor symbol you use.
It works great for the res_xhigh_po resistors as shown in the picture.

RTimothyEdwards, 2023-10-24:
@d-m-bailey :  While it's true that this is just an estimate, the difference between 9.1kOhms and 31.8kOhms is way too much to be accounted for by, say, lack of terminal resistance.

RTimothyEdwards, 2023-10-24:
@d-m-bailey , @mhommelga :  The error is in the symbol definitions for the specific-width devices.  For example,
`res_high_po_5p73.sym` has
```
tcleval(R=[ev \{ 56.46 * @L / 5.73 / @mult \}])
```
The error is that the width of 5.73um is _already_ folded into the value 56.46 and should not be divided out again.  I will inform Stefan Schippers of the error.

StefanSchippers, 2023-10-24:
I have pushed the fix [upstream](https://github.com/StefanSchippers/xschem_sky130).
The displayed value is now a better estimate since the value is split into a fixed (contact) resistance and a sheet (per square) resistance.
[image]

mhommelga, 2023-10-25:
Thanks for the quick turn around and resolution.

closed 2023-10-25 by mhommelga