fossi-foundation/open-pdks #413: Weird cell LEF definitions for sky130

donn, 2023-11-01:
1. All EF cells
  * Lack a `SITE` definition
  * Are improperly(?) marked `CLASS BLOCK` instead of `CLASS CORE` as normal standard cells are

```lef
MACRO sky130_ef_sc_hd__decap_12
  CLASS BLOCK ;
  FOREIGN sky130_ef_sc_hd__decap_12 ;
  ORIGIN 0.000 0.000 ;
  SIZE 5.520 BY 2.720 ;
  PIN VGND
  …
```

2. All double-height cells in at least `fd_sc_hd`
  * Have their sites declared as `unithd` instead of the correct `unithddbl`, e.g.

```lef
MACRO sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1
  CLASS CORE WELLTAP ;
  FOREIGN sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 ;
  ORIGIN 0.000 0.000 ;
  SIZE 6.440 BY 5.440 ;
  SYMMETRY X Y R90 ;
  SITE unithd ;
  …
```

RTimothyEdwards, 2023-11-02:
For the cells in (1), fix them the way you like in `sky130/custom/sky130_fd_sc_hd/lef` and send me a pull request.
For the cells in (2), fix them in the https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd repository and send a pull request.

donn, 2023-11-02:
https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd/pull/12
https://github.com/RTimothyEdwards/open_pdks/pull/414

closed 2023-11-02 by donn