fossi-foundation/open-pdks #415: Getting Errors in building

rudro101, 2023-11-07:
I have followed the steps to build open_pdks. ./configure is also done successfully. But in make step, getting this error.
[image]
I have tried multiple times but each time same error.
I am using Ubuntu22.04 distro in WSL2.

RTimothyEdwards, 2023-11-08:
This is a bit odd because normally there isn't an error in a stock Ubuntu build;  however, it being WSL2 there might be discrepancies that pop up.  The error will not be immediately obvious from the "make" output and you will probably need to look into the log file, which can be difficult to interpret because the various tools and scripts generate a lot of non-fatal warnings and error messages, and which can also be difficult to interpret because output from multiple threads get scrambled.  And which tends to be a very large file.  However, if you could post a compressed version of it, that would be helpful.  You could also try posting the log file to me in the open-source-silicon slack workspace.

rudro101, 2023-11-08:
Thank you so much for your response. However, I noticed that my magic version was old. So updated that and the problem is solved. Don't know the logic

rudro101, 2023-11-08:
> This is a bit odd because normally there isn't an error in a stock Ubuntu build; however, it being WSL2 there might be discrepancies that pop up. The error will not be immediately obvious from the [...]

Thank you so much for your response. However, I noticed that my magic version was old. So updated that and the problem is solved. Don't know the logic

closed 2023-11-08 by rudro101