fossi-foundation/open-pdks #427: Make gives me several errors regarding LEF, CDL, Magic etc.. Cell might have been instanced before they were defined

MatiasJHZ, 2024-01-20:
I tried to run make on open_pdk but I am getting several errors.

I made sure I have all prerequisites according to the instructions on open_pdks webpage.

Here is my sky130A_make.log file to show you my errors, they are several.
[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/13998624/sky130A_make.log)

These are some errors type examples:
`Error message output from magic script:
Couldn't find label VPB
No port found with name VPB.
Can't write file named '(UNNAMED)'
Annotating files in /home/matiasjhz/open_pdks/sky130/sky130B/libs.ref/sky130_fd_sc_lp/maglef
make[3]: Leaving directory '/home/matiasjhz/open_pdks/sky130'
make[3]: Leaving directory '/home/matiasjhz/open_pdks/sky130'`

`Warning:  cell L1M1_CDNS_52468879185306 already existed before reading GDS!

Error message output from magic:
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "sky130_fd_io__overlay_vddio_lvc" (byte position 735778): Boundary is not closed.
Error while reading cell "sky130_fd_io__res250_sub_small" (byte position 328716): NODE elements not supported: skipping.`

`Error while reading cell "sky130_fd_io__gpiovrefv2_hv_nor2" (byte position 101846): Warning: cell "nfet_CDNS_52468879185814" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_ctl_lsbank" (byte position 263124): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_ctl" (byte position 303008): Warning: cell "M1M2_CDNS_52468879185967" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_hotswap_latch" (byte position 3551354): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_res_weak" (byte position 5785268): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_ctl_lsbank" (byte position 7076664): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_ctl" (byte position 7116548): Warning: cell "M1M2_CDNS_52468879185967" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_hotswap_latch" (byte position 10785212): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_res_weak" (byte position 16804356): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.`

`Error while reading cell "sky130_fd_io__sio_ctl_lsbank" (byte position 85274120): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_ctl" (byte position 85435412): Warning: cell "M1M2_CDNS_52468879185967" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_hotswap_latch" (byte position 88598650): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_fd_io__sio_res_weak" (byte position 90832074): Warning: cell "L1M1_CDNS_5246887918527" placed on top of itself.  Ignoring the extra one.
Error while reading cell "sky130_ef_io__gpiov2_pad_wrapped" (byte position 23648): Cell "sky130_ef_io__gpiov2_pad" is used but not defined in this file.
Error while reading cell "sky130_ef_io__gpiov2_pad_wrapped" (byte position 23648): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad_wrapped" (byte position 23648): cell sky130_ef_io__gpiov2_pad was used but not defined.
Error while reading cell "sky130_ef_io__simple_pad_and_busses" (byte position 9381712): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__com_pg_esd" (byte position 9381712): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vccd_lvc_clamped3_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssd_lvc_clamped3_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vddio_lvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssio_lvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vddio_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vddio_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssio_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssio_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vccd_lvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vdda_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssa_hvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssd_lvc_clamped_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssd_hvc_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vssa_lvc_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__corner_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vdda_lvc_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__vccd_hvc_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Error while reading cell "sky130_ef_io__gpiov2_pad" (byte position 3852674): Rewinding input.  Cells may have been instanced before they were defined.  Consider using "gds ordering on".
Couldn't find label AMUXBUS_A
Couldn't find label AMUXBUS_B
Couldn't find label VSSA
Couldn't find label VDDA
Couldn't find label VSWITCH
Couldn't find label VDDIO_Q
Couldn't find label VCCHIB
Couldn't find label VDDIO
Couldn't find label VCCD
[... 64 lines cut ...]
No CDL file contains sky130_fd_io device sky130_fd_io__top_power_hvc_wpad
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vssd_lvc
No CDL file contains sky130_fd_io device sky130_fd_io__top_gpiovrefv2
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vssa_hvc
[... cut ...]
No CDL file contains sky130_fd_io device sky130_fd_io__overlay_vccd_hvc
No CDL file contains sky130_fd_io device sky130_fd_io__top_hvclamp`

I would appreciate any help. I don't know much about linux, make, scripts, bash and all that stuff. Main reason why I like hardware more than software. This is really overwhelming regarding the things that it does behind the curtain and I don't know what I did wrong or if I am missing something.

So far I've been following these instructions: [https://www.youtube.com/watch?v=KgBLByOkJxA](url)

RTimothyEdwards, 2024-01-20:
It looks like you had a successful build.
I cannot control all of the various warning and error messages produced by the multitude of tools used by the open_pdks installer, and I don't have the time to track them all down and try to filter them out.  If it didn't halt mid-build with a _fatal_ error and a returned error code, then it's fine.

MatiasJHZ, 2024-01-20:
ok, thank you! I will proceed with the installation. I have some simple questions though:

1) I git cloned skywater-pdk, magic and open_pdks on my /home/usr folder. Some errors are about not being able to find some files. I am quite sure I build succesfully and with no errors Magic. How can I make sure there is no path errors or directory locations that I shouldn't be using? or the installation is agnostic with respect to where those files are located as long as you provide the right path? Could have been the fact that I downloaded first the skywater pdk before building and installing open-pdk (as suggested in this [https://www.youtube.com/watch?v=KgBLByOkJxA&t=0s](url) youtube video) when open-pdk installation takes care of the download of the pdk by itself?

2) First I run: `./configure --enable-sky130-pdk=/home/usr/skywater-pdk --prefix=/home/usr `, then `sudo make` and I got the errors I showed in this Issue. Then I tried the "impatient way" without providing the path, I just typed: `./configure --enable-sky130-pdk ` and then `sudo make`, which yielded errors again. I am sorry for the stupid question, I know nothing about building and make, but after a configure and make, If I want to run another configure and make, do I have to clean up the previous configure and make? or the last process takes precedence? 

Thank you!

RTimothyEdwards, 2024-01-21:
(1) I no longer use the top-level path to the PDK, because it is more convenient to work with the individual libraries separately.  However, that might mean that "--enable-sky130-pdk=<path>" does not actually record the <path> but only enables the PDK for installation, in which case my instructions are wrong and need to be corrected.  The videos are correct for the version of open_pdks at the time they were made, but are now out of date.  Given that the top-level path only exists in the google repository, and only a few libraries are still pulled from the original google repositories (as opposed to the patched efabless forks, which I maintain and update myself), I need to deprecate the use of a top-level source path and flag it as an error.

(2)  I strongly discourage using any user directory path as the install path, since that makes the whole PDK user-writeable, which is just asking for trouble.  At any rate, the value of "prefix" has absolutely no effect on the build until you do "make install", at which point the entire PDK is copied from the staging area in open_pdks to the location specified by "prefix".

I suggest running "make veryclean" between builds.  I believe that the build will not be affected by files that already exist in the staging area when "make" is run, but I'm not certain.

MatiasJHZ, 2024-01-21:
Thank you for your reply.

I am confused though. When you say your instructions are wrong, you meant your instructions on opencircuitdesign.com or in github? For the latest version of open_pdks, which instructions should I follow?

PS: You were right, I was able to run `sudo make install` and the installation went with no errors. So as long as there are no fatal errors during `make,` one could proceed with the installation.

RTimothyEdwards, 2024-01-22:
The instructions are wrong everywhere they suggest that `--enable-sky130-pdk=<path>` is a valid configuration;  in fact, that configuration is equivalent to `--enable-sky130-pdk=yes` or `--enable-sky130-pdk` and `<path>` is ignored completely.